vitotai/BrewPiLess #96: Authenticate apply operation

guma44, 2018-06-01:
Hi,

I was searching but I could not find it in the options. Is it possible to authenticate apply operation? I would like to see my fermentation "on the internet" but it seems that apply operation cannot be restricted to "logged-in user". Also, is it possible to switch to https? 

Rafal

allthepies, 2018-06-01:
You could modify the s/w to require authentication prior to changing the BPL mode.
https is beyond the capabilities of the ESP8266.

guma44, 2018-06-01:
Thanks, I thought so. I will look at the code.

vitotai, 2018-06-02:
Well, the apply should already be protected if there is no bugs...

HTTPS seems not possible for ESP8266, which as too limited memory to run SSL AND other jobs.

doliver-pulsemining, 2018-06-05:
Maybe run it through cloudflare or similar if you want https?

guma44, 2018-06-05:
I will check it today again, although I have a lot of problems when restarting it - sometimes it just will not generate any website ie. the ip is not accessible although I see it connected to the network. That is a different topic anyway.