vitotai/BrewPiLess #98: keep last iSpindel Temperature

ollinator2000, 2018-06-10:
Hi,

i m nearly done with my whole brewpiless setup . ( i ll show later on when its done ) 
what i noticed, when connecting iSpindel to brewpiless the measured temperature will only be displayed for a short time period. 
when Spindel is configured to send data only every 15 minutes , then 2 minutes the temperature will be displayed and 13 minutes ( roughly guessed ;-) ) not. 
maybe it s possible to display the last measured data until new comes in . 
i m not very familiar with C Code , i m more of a python guy, so i unfortunately i can't code it myself. 

Its not a bug , more a feature request... ;-) 

Regards 
Oliver

vitotai, 2018-06-11:
I don't really understand you.
The temperature from iSpindel is recorded as "Aux Temp". I never thought of a way that it could disappear.

ollinator2000, 2018-06-13:
Sorry , 
ill try to explain. The Temperature send by iSpindel is only shown aprox 1 or 2 minutes on the display. 
i think that has something to do with the fact that my Spindel sends its data only every 15 minutes. only the first 2 minutes after the Spindel sends its data the temperature is visible . after that the display will show nothing behind Beer Temperature. 

So the Beer Temperatur will only shown for a short period of time it has been send by iSpindel.
If you log into brewpiless by browser you can look at the AUX and Beer Graph of course, but i would be very cool if the last temperature is shown on display . ;-) 

I uploaded 2 images which shows both states. 

Temperature shown on Display
https://cloudy.van-der-velde.net/index.php/s/99SDf98jmGFEwaJ

Temperature not shown on Display
https://cloudy.van-der-velde.net/index.php/s/sTPHACisNgE3dTC

vitotai, 2018-06-14:
Well, I see now.
If you want to use iSpindel as beer sensor, the report period should be shorter. The default duration to invalidate the temperature should be 300 seconds. Let me dig it up why only 2 minutes.

If you want change it, you can modify line 13 of TempSensorWireless.h
`
WirelessTempSensor(bool connected=false,fixed4_4 cal=0,uint32_t expiryTime=300){
`
Change the 300(seconds) to whatever you want. However, I don't know what you will get by using this long duration.

vitotai, 2018-07-20:
https://github.com/vitotai/BrewPiLess/wiki/Software-Installation