vitotai/BrewPiLess #100: Non-intuitive names for temperature constants on display

Paiakan, 2018-07-06:
It always made me confused the display showing the constants "Beer" and "Const." meaning "Real/Actual Temperature" and "Set Temperature". This layout also implies the word "Beer" appearing twice.

And maybe the first word "Mode" could be replaced by a word to represent the current cooling method "Beer Const", "Fridge Const" or "Beer Profile". It could look like this:

Beer C. Actual Set
Beer 20.2 20.0
Fridge 18.3 18.0
Cooling for xxminyy

vitotai, 2018-07-06:
First of all, the display is exactly what original BrewPi displays. I think most people get used to it. 
Second, the "Mode" string is an input item of the menu when using buttons or rotary encoder. It is indeed redundant but changing it involves more than what you have thought.
Third, the interface created by Tom relies on the specific LCD strings to get the current information displayed on the LCD. It can be changed, but I don't feel like so for now. Although I might change it later.

billy-bedlam, 2018-10-10:
I know this is digging up an old issue,  but if there was to be a change made in the future I think another way to make it more obvious would be just adding PV (Process Variable) and SP (Setpoint) prefixes something like:

Mode   Beer Const. 
Beer   PV 20.2   SP 20.0
Fridge   PV 18.3   SP 18.0
Cooling for   xxminyy

Might be squished on a 4x20 LCD though.

They are standard abbreviations in control systems / process control industries and seen on pretty much every PID controller.

vitotai, 2018-10-10:
Again. In my opinion, it's more like a preference thing. You can do it if you want, but such modification will break Tom's frontend.