vitotai/BrewPiLess #101: default wifi password not working

brendongough, 2018-07-08:
Attempted to use the latest 2.5 and 3.0rc pre-built images, standard and thorrax.
I have flashed these with esptools and nodemcu without any luck. I've also used platformio.
I have also attempted to erase flash without any success.
I have tested two separate boards with the same results.
I am unable to use the password 'brewpiless' or 'password' with the default network on my phone, Mac or Windows computer on either of the two boards I have tested and various methods of flashing.
The boards I am using are WeMos D1 mini PROs, they have no previously had any firmware flashed to them.
I have attempted to use Putty and Termite with baud 115200 in an attempt to get the wifi password in the debug message without success. 
I have also attempted various other baud rates however I can see I should be testing with 115200. 
Part of the console message I do receive is: rl[00]lœž|[00]Œlà|[02][0C][0C][0C]

I'm not sure where to go from here as my experience with the platform and project is quite limited however have attempted to research and troubleshoot as much as possible before raising the issue.

Thank you in advance.

vitotai, 2018-07-09:
Did you see the network "brewpiless"?

Please try flash the D1 SOLO without anything attached except the USB cable. You might need to verify by using other sketch.

brendongough, 2018-07-10:
Just tested a new / blank board with various pre-compiled bins from here as well as using platformio with no success.
I flashed brewpi-esp8266 and iSpindel with no issue, also on boards which are fully soldered up.
Any ideas?

vitotai, 2018-07-10:
If you can flashed brewpi-esp8266, then you should be able to connect it to your router. After that, the WiFi credential will be stored even after re-flashed. In this case, you won't find "brewpiless" network.

brendongough, 2018-09-03:
I haven't had any success with this at all. I can get brewpi-esp8266 connected to the wifi without an issue however when I flash brewpiless over the top, it seems the wifi config may not be saved or read from spiffs correctly? What can I do/run to keep diagnosing?

brendongough, 2018-09-03:
I've tried an erase, flash with https://github.com/tzapu/WiFiManager and successfully configure with the wifi. Once this was successful I slashed brewpiless over the top in a hope that the same wifi library in use has been used and hopefully writes out the wifi config to spiffs correctly. No luck :(

vitotai, 2018-09-04:
The WiFi credential is not stored on SPIFFS. It is stored in a reserved area on the flash.
What stored on the SPIFFS is the working mode: AP only, AP and STATION, or STATION ONLY. You might need to check this setting if you can.

brendongough, 2018-09-04:
I assume this issue is because of the ESP8266 I'm using, however it is weird that they work fine with iSpindel and brewpi-esp8266, Wemos Mini D1 Pro below:
[image]

I'm happy to build from a branch if needed, is there anything which you/I can do to remove the password from the AP?

Do you feel there would be any benefit of upgrading tzapu to v0.14?

vitotai, 2018-09-04:
That is a D1 pro with 8M flash. I have one which works fine.

I still know what is your problem.

If the board used has connected, it would connect the the same network without creating a brewpiless network. You have to manually setup to make it create a brewpiless. If you can't even connect to it, there should be no brewpiless.

You can modify the code to never use password:
https://github.com/vitotai/BrewPiLess/blob/7afc58f9781dd9c5a4760a89b3b7909977756866/src/WiFiSetup.cpp#L80
change
`	bool connected= wifiManager.startConfigPortal(_apName,_apPassword);`
  to
`	bool connected= wifiManager.startConfigPortal(_apName);`

brendongough, 2018-09-04:
Thanks, I've got the AP starting up without a password however cannot connect from my phone, I'll try some devices later tonight. After the initial timeout the AP does set a password which is expected, I still can't connect using brewpiless/brewpiless however.

Is there any way the sketch can haveh hard coded network information somehow so the file is written to on start? It would be great to have a sketch which doesn't rely on this part of the setup process which seems to be problematic.

vitotai, 2018-09-04:
If that is what you want. 
Just modify the 
https://github.com/vitotai/BrewPiLess/blob/7afc58f9781dd9c5a4760a89b3b7909977756866/src/WiFiSetup.cpp#L175

Change the line:
`startWiFiManager(false);`
to
`WiFi.begin("SSID","password");`

Use real SSID and password. 

That code is indeed problematic, and I am planing to change it. However, I don't think that would solve your problem.

brendongough, 2018-09-04:
I can confirm setting my network SSID and password at line 175 as you described worked perfectly straight up 👍

brendongough, 2018-10-11:
I tried out 3.1 hoping this would be fixed but it seems it does the same thing. As per #123 I used the 16m flash setting to build with no luck. When I attempt to use brewpiless/brewpiless user/pass I still get password failed.

Since the wifi manager has changed in 3.1, the above fix isn't available. Is there any way I can do the same in 3.1?

vitotai, 2018-10-12:
To connect BPL to your router:
in WiFiSetup.cpp line 65
https://github.com/vitotai/BrewPiLess/blob/84e140334b315720ab93e72570cf359be4daf09d/src/WiFiSetup.cpp#L65

change 
``````
		WiFi.begin();
``````

``````
		WiFi.begin("YOUR SSID","YOUR PASSWD");
``````

To use a predefined username/password:
line 67:
https://github.com/vitotai/BrewPiLess/blob/84e140334b315720ab93e72570cf359be4daf09d/src/WiFiSetup.cpp#L67

``````
WiFi.softAP(_apName, _apPassword);
``````
to
``````
WiFi.softAP(_apName, "brewpiless");
``````
 I would suggest you change the "board" setting in platformio.ini. It might need to be "d1_mini_pro".

http://docs.platformio.org/en/latest/boards/espressif8266/d1_mini_pro.html#board-espressif8266-d1-mini-pro

brendongough, 2018-10-12:
@vitotai seems the board setting doesn't play nice with 1.5, 1.6, 1.7 or default, results in a tonne of errors. What happens if you try and build this with 'platform = espressif8266@~1.6' etc ? I'm unsure however it certainly seems there's some dependencies on using 1.5. I've tried to use the d1 pro setting and the normal nodemcuv2.

The wifi configuration suggestion works just fine however there's a few other weird issues with crashing which I assume has to do with the geometry of the flash and/or the board settings. I have a few normal non-pro wemos arriving shortly. I wrongfully assumed the pros were quite interchangeable with the normal wemos as it seems there's a fair bit of work involved.

vitotai, 2018-10-13:
Long story short. The program size would be too large for OTA update with 1.6. Ditching WiFiManager after v3.1, the program size is smaller. Therefore, we can do it with framework 1.8.

Try this configuration:
``````
[env:d1pro]
platform = espressif8266@~1.8.0
board = nodemcuv2
framework = arduino
lib_extra_dirs = ./lib_async
build_flags = -Wl,-Tesp8266.flash.8m.ld
``````

BTW, if you can access http://brewpiless.local:8008/filemanager, please download the "bpl.cfg" file and share with me for further investigation.

brendongough, 2018-10-14:
@vitotai Using your d1pro env profile, everything seems to work including the initial wifi setup which is great - thanks very much :)

bpl.cfg attached below as a .zip as requested.
[bpl.zip](https://github.com/vitotai/BrewPiLess/files/2476146/bpl.zip)

 I'm using 'DBOARD=Thorrak_PCB" successfully as well. Curious, is "DLegacyEspAsyncLibraries=true" required for anything in particular?

Whilst watching the board on my desk I did see it randomly reboot. I'll continue to monitor stability tonight and see what happens. I'll try using a RasPi 2.5A USB adapter as well just in case it is a power issue.

vitotai, 2018-10-14:
Yesterday, I tried framework 1.8 and found it reboot 3 times in 12 hours, which is unacceptable.
I am going to stick with 1.5 for a while.

LegacyEspAsyncLibraries is used for older version of ESPAsyncWebServer which works with older version of framework.

The configuration file seems OK. I have no idea where went wrong.

brendongough, 2018-10-14:
@vitotai I had a go with 1.6.0 which is the first version which seems to introduce the board type, unfortunately I was faced with wireless connectivity issues much like with 1.5. 

I assume a fair bit of the core needs changing to work with 1.8.0 :( At this stage I think I'll need to just use the normal D1 boards to avoid any issues, what do you think? Is it worth suggesting against people using the pro boards?

Anything above 1.6 compiles with the following warnings (yellow, not read), the below is example is for 1.7.3

```
.piolibdeps\OneWire_ID1\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
.piolibdeps\OneWire_ID1\OneWire.cpp:167:24: warning: unused variable 'reg' [-Wunused-variable]
volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
^
.piolibdeps\OneWire_ID1\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
.piolibdeps\OneWire_ID1\OneWire.cpp:201:24: warning: unused variable 'reg' [-Wunused-variable]
volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
^
.piolibdeps\OneWire_ID1\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
.piolibdeps\OneWire_ID1\OneWire.cpp:229:24: warning: unused variable 'reg' [-Wunused-variable]
volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
^

In file included from lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp:28:0:
lib\esp8266-oled-ssd1306-master\OLEDDisplay.h: In member function 'virtual bool OLEDDisplay::connect()':
lib\esp8266-oled-ssd1306-master\OLEDDisplay.h:253:29: warning: no return statement in function returning non-void [-Wreturn-type]
virtual bool connect() {};
^
lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp: In member function 'void OLEDDisplay::drawStringInternal(int16_t, int16_t, char*, uint16_t, uint16_t)':
lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp:381:10: warning: enumeration value 'TEXT_ALIGN_LEFT' not handled in switch [-Wswitch]
switch (textAlignment) {
^
lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp: In member function 'void OLEDDisplay::drawInternal(int16_t, int16_t, int16_t, int16_t, const char*, uint16_t, uint16_t)':
lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp:722:13: warning: unused variable 'yScreenPos' [-Wunused-variable]
int16_t yScreenPos = yMove + yOffset;
^

lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp: In member function 'void OLEDDisplay::drawXbm(int16_t, int16_t, int16_t, int16_t, const char*)':
lib\esp8266-oled-ssd1306-master\OLEDDisplay.cpp:361:19: warning: 'data' may be used uninitialized in this function [-Wmaybe-uninitialized]
data >>= 1; // Move a bit
^

In file included from lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.h:31:0,
from lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:27:
lib\esp8266-oled-ssd1306-master\OLEDDisplay.h: In member function 'virtual bool OLEDDisplay::connect()':
lib\esp8266-oled-ssd1306-master\OLEDDisplay.h:253:29: warning: no return statement in function returning non-void [-Wreturn-type]
virtual bool connect() {};
^

lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp: In member function 'void OLEDDisplayUi::drawFrame()':
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:280:48: warning: 'y1' may be used uninitialized in this function [-Wmaybe-uninitialized]
x *= dir; y *= dir; x1 *= dir; y1 *= dir;
^
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:280:37: warning: 'x1' may be used uninitialized in this function [-Wmaybe-uninitialized]
x *= dir; y *= dir; x1 *= dir; y1 *= dir;
^
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:280:26: warning: 'y' may be used uninitialized in this function [-Wmaybe-uninitialized]
x *= dir; y *= dir; x1 *= dir; y1 *= dir;
^
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:280:16: warning: 'x' may be used uninitialized in this function [-Wmaybe-uninitialized]
x *= dir; y *= dir; x1 *= dir; y1 *= dir;
^
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp: In member function 'void OLEDDisplayUi::drawIndicator()':
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:389:53: warning: 'y' may be used uninitialized in this function [-Wmaybe-uninitialized]
this->display->drawFastImage(x, y, 8, 8, image);
^
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:389:53: warning: 'x' may be used uninitialized in this function [-Wmaybe-uninitialized]
lib\esp8266-oled-ssd1306-master\OLEDDisplayUi.cpp:383:7: warning: 'posOfHighlightFrame' may be used uninitialized in this function [-Wmaybe-uninitialized]
if (posOfHighlightFrame == i) {
^
```

vitotai, 2018-10-15:
My d1 mini pro works just fine with nodemcu2 setting. 
The problem is that we never know if it is genuine one or knock-off. 

The warnings are in sources of libraries. It should be OK, and I preferred leave it as is.