vitotai/BrewPiLess #102: Simple question on control

RollyBrewer, 2018-07-10:
Hey - 

first off, this project is AWESOME!  I have looked at BrewPi for a while but it's too complex for my taste...this project is exactly what i was looking for - i THINK.  Just need one question answered:

Is the control method simple on/off of the relays, with some sort of pseudo PID control?  You can't apply variable heat with the mechanical relays shown in the hardware setup example, and of course you can't with the cooling anyway.  So is it like the PID predicts when to turn the heating or cooling off before the fridge temp over or undershoots the desired temp?

I am wanting to upgrade my setup from a STC-1000 type controller, but seeing as this project still uses a relay for heating and cooling, what i describe above must be the enhancement. Of course the logging and web and LCD UI is an improvement too! 

Is my hopeful thinking correct, in that it is as simple as flashing the default .bin file onto my Wemos D1, and my existing fridge suddenly gets much better control and logging???

vitotai, 2018-07-10:
First of all, the temperature control algorithm and source code come from the original BrewPi, or legacy version of BrewPi. You might find more information by googling BrewPi.
My understanding of the algorithm follows.
One of the design goal, the most important one IMO, is control BEER temperature by controlling FRIDGE temperature. The INPUT of PID is BEER temperature while the OUTPUT is DESIRED FRIDGE temperature, aka FRIDGE SET. (No PID is applied in Fridge mode, so I just ignore that.). I am no expert of PID, so a simple example is the best I can do to explain. For example, if the desired beer temperature is 20C/68F, 
if current Beer temp reading is 30C/86F, the output of PID will be a very low value, like 1C/34F.
if current Beer temp reading is 21C/70F, the output of PID should be a very close value, like 19C/67F. 
The algorithm constantly monitor the beer temperature and set the desired fridge temperature. When the fridge temperature meets the desired value, the controller turns off the fridge, or the compressor to be exact.
In addition to PID, the algorithm also takes hysteresis and minimum compressor operation/idle time into consideration.

Most of the parameters can be manual changed. Please check
https://github.com/vitotai/BrewPiLess/wiki/JSON-command

Heads-up: the default value works pretty for most people who use ONLY fridge AND thermowell for beer temperature. The parameters must be tuned to get good result when HEATING is applied, given  the fact that most people use TOO powerful heating devices.

RollyBrewer, 2018-07-10:
Thanks - can you help with where to edit the paramaters you mention for heating?  I am reading the whole wiki now as well to try and help myself :-)

RollyBrewer, 2018-07-11:
Great, thank you very much, I appreciate the help!  I have v3rc6 flashed, running, and now am looking at those tuning values :-)  I understand about a minimum of say 5 mins for the compressor health.

I am still interestd in how well this will work compared to the STC-1000 because although it has PID algortithms where the STC-1000 is just ON/OFF hysteresis, even BPL can still only turn the relay on or off, there is no proportional control so i guess it must turn off heating or cooling early, knowing that the beer will continue to heat or cool a bit.  I will enjoy running some tests with buckets of water ;-)

vitotai, 2018-07-11:
I don't know much about STC-1000, but I bet that if there is something that can do better than BrewPi, then it must be the latest version of BrewPi. In terms of cooling during fermentation, it should be par. The latest spark BrewPi should be better in heating control.
One thing to bear in mind: "Garbage in, garbage out". A thermowell to measure beer temperature is the way to go.

RollyBrewer, 2018-07-11:
Well, I have been having an entertaining time with this - and yes I will be using thermowell.

I think i know that answer to this next thing, but why do i see:

"Free Space: -1,073,740,520 Bytes, Used Space: 1,073,740,536 Bytes, Total Space: 16 Bytes"
I think it is because i am using an ESP-01 with 4Mbit/1MByte flash, is that correct?  I have a Wemos D1 R2 ordered for the real setup, I am only using the ESP-01 for now since i had it already.

vitotai, 2018-07-11:
The default configuration requires 4M Bytes. You will have to use Sonoff configuration if you don't want to build by yourself.