vitotai/BrewPiLess #105: iSpindel gravity is not displayed

peteng72, 2018-07-18:
Hello,
I have successfully connected my iSpindel to my BPL v3.
The tilt, temp and battery values are displayed correctly, but not the gravity. It doesn’t matter whether I choose ”calculated by BPL”, or not.
Have I missed something?
Rgds,
Peter

vitotai, 2018-07-18:
If "calculated by BPL" is NOT enabled, BPL looks at "gravity" field from iSpindel.
If "calculated by BPL" is enabled, the coefficients must be supplied, and the "gravity" from iSpindel is ignored.

There is an additional "calibrate iSpindel" setting when logging is started. If that is enabled, all the gravity is calculated by Web Browser, and you will have to input gravity data during fermentation.

peteng72, 2018-07-18:
Settings:

[image]

Result:

[image]

ollinator2000, 2018-07-18:
i coded some commands for testing my iSpindel TCP Server -> brewpiless Adapter . you can use one of the to simulate Spindel Data. ( Makes everything much easier ) May be you can use it. 

curl -vv  -A "iSpindel" -H "Content-Type: application/json" -X POST -d '{"name":"iSpindel","angle":28.67005,"temperature":27.3125,"battery":4.19708,"gravity":1.04
}' http://brewpiless.local:80/gravity

peteng72, 2018-07-18:
Thanks @ollinator2000 , that helped a lot!
Your command worked fine, and I was able to find out that my iSpindel's gravity values were in Plato.
BPL doesn't accept gravity values outside the interval 0.80<gravity<1.25, so the default SG=-1.000 reading was never replaced.
I was previously using it this way together with Brewfather, but now I need to recalculate the coefficients for SG. Will do a "Brew and Calibrate".

ollinator2000, 2018-07-19:
Really ??? I m not at and can't check but I m pretty sure my came as gravity value .

Regards
Olli