vitotai/BrewPiLess #106: Temperature correction of iSpindel

peteng72, 2018-07-22:
I just performed a manual calibration of my iSpindel at 20 degree C. It works as expected.
However, the temperature drift of the accelerometer is huge.
Example:

Temp (C)    iSpindel (SG)    Hydrometer (SG)
20                1.000                 1.000
5                  0.988                 1.003

This needs to be corrected for, since the fermentation temperature could vary between 5-22 degrees.
I think it’s possible to include the variable ”temp” in the iSpindel polynom, but it would be better to do it iBPL. 
Would it be possible to implement support for this?

vitotai, 2018-07-23:
There is already an option for temperature correction, just under the coefficients setting.

What is your formula for temperature correction? I check your data on brewersfriend.com, it seems that 1.001@5C will be corrected to 1.000@20C and 1.003@5C will be corrected to 1.002@20C.

peteng72, 2018-07-28:
There are two different physical processes that needs to be temperature corrected:

1. The wort density changes with temperature. 
    This is corrected with the option you mention above and is the same for all iSpindels.

2. The accelerometer (MPU 6050) changes with temperature.
     This temperature drift is unique for each iSpindel and could be corrected with something
     like this:

     SG – (TemperatureCorrectionFactor X (Temperature – 20 °C))

    Have a look at the discussion below:
    https://github.com/universam1/iSpindel/issues/6

    They have solved it by having a polynom with the variables ”tilt” and ”temp”.
    The polynom could look like:

    0.005 * (tilt)(tilt)(tilt) - 0.004 * (tilt)(tilt) + 0.003 * (tilt) + 0.5 * ((temp) - 20) + 0.95

vitotai, 2018-07-28:
Well.
I thought I've read some discussion about accelerometer temperature drift, but I don't follow it and wasn't aware of the final solution.

How do you get the TemperatureCorrectionFactor ?

peteng72, 2018-07-28:
The iSpindel could be placed in a fixed position in the fermentation chamber.
Pairs of temperature and tilt (or SG) could then be plotted and curve fitted, similarly to the polynom coefficients.