vitotai/BrewPiLess #107: Cooling Time

VictorSRodrigues, 2018-07-27:
I'm having this problem:
I use BPL in a freezer.
When the temperature goes up, BPL turns on the fridge.
That's right.
The question is that the freezer stays "ON" just for some minutes... 1~2 minuts... Never more that 3 minutes.
This action could be dangerous to the compressor, right?
The "minCoolTime" parameter is used to keep the fridge ON for some minimum time? Cause I've seted it to 37856 (I think is the maximum value) but BPL keeps turning OFF the fridge in less than 3 minutes.
Am I doing somethig wrong?

BIN file is "BrewPiLess.v2.6rc.oled" (but I had the same problem with 3.0).
I use rele modules.

Thanks!

vitotai, 2018-07-28:
It might be a sensor disconnection issue.
The only condition that minimum cooling time is not respected is when the sensors disconnected.

VictorSRodrigues, 2018-07-28:
I got it.
I'll rewire all my system and try again.
Thank for your attention.

And, like BMEx, this project is amazing!

VictorSRodrigues, 2018-07-31:
Hello again.
Vito, by the code, there is a specific sensor whose disconnection lead to this issue or any sensor disconnection causes this?
Cause I have rewired all my system and I had tried to test all the sensors with a simple temperature monitoring script but I can't find any problem and the issue keeps ocurring.
May be you can help me....

Thanks again.

vitotai, 2018-09-20:
This is the "core part" that comes directly from BrewPi, which I tried not to touch.
The original BrewPi has debug message for sensor disconnection. Here is a way to check, but I am not sure if it works.
open testcmd.htm, like http://brewpiless.local/tetcmd.htm

Then , just wait unlit the issue happens. Check if any message starts with "D", which stands for Debug. If it is sensor issue, you might see the sensor address in that message.