vitotai/BrewPiLess #110: LCD Issue

dotps1, 2018-08-28:
Last night my LCD went all crazy, i had to turn the unit off and back on.  Not sure what i can do to give more info on it.
[image]

vitotai, 2018-08-29:
It might be EMI or bad connection.
There is a LCD refresh action every 10 minutes. If it were EMI, that refresh should fix it.

dotps1, 2018-08-31:
yeah im not sure, it seems to run fine for a day or two, but then does it again.  i let it sit for atleast an hour after i noticed it again, and never refreshed.

anibalbenedetto, 2018-08-31:
Hello Thomas, its seems to be electric noise. You can try changing the
power supply.

I had a similar issue with other proyect and i solve it adding a tantalum
capacitor to de power line in the display.

Anibal

dotps1, 2018-08-31:
i cannot do it very easily, i have everything running off an inline AC>DC 5v converter.  You can probably see it in the pictures here:  https://www.thingiverse.com/thing:3042974  But its a 5v display running off a 5v power supply, so i'm not sure where the noise would be coming from.

dotps1, 2018-09-01:
its on the far side of the box from the actual plug into the LCD.  Its strange it will usually work for a few days before it goes wonky like this.  I have not measured the ripple when the load changes, i am kinda new to making my own stuff like this so i'm not sure even how to do that.  Also, is there a way i could shield it?

quantum8, 2018-09-06:
Just make sure the wire from the LCD is completely clear of the relays, as well as the wires coming from the relay. You could try shielding; you'd need to replace the wires from the wemos to the LCD with a screened multicore cable. Make sure the shield is only connected to ground at one end otherwise you'll induce more noise instead. This is probably overkill here though.

You'll only be able to measure ripple if you have a oscilloscope and happen to catch it happening as it seems to be transient. Is there something nearby that is causing it to screw up? A microwave, amplifier or something nearby? It could even be your fridge causing interference.

As  anibalbenedetto suggested, you could try a capacitor on the output of the power supply. A basic summary on the workings of this is here: https://electronics.stackexchange.com/questions/21686/whats-the-purpose-of-two-capacitors-in-parallel

dotps1, 2018-09-17:
So, I guess Im going to go ahead and close this.  I'm about 9 days into a brew, and have even been using a profile ramping up the temp over the past 3 days, and this has not occurred.  And with the ramp up, its defiantly turning on and off more.  Anyway, I can't explain it, and I haven't changed anything.