vitotai/BrewPiLess #123: Not setting up the network

filipematos95, 2018-09-25:
Hello, 
I just installed BPL on my wemos d1 mini and when I try to set up the wifi I just get a ap with the username: brewpiless but also with a password:brewpiless so I am not able to open the set up page

vitotai, 2018-09-26:
Connect to brewpiless, and use your browser to connect to http://192.168.4.1 if the setup page doesn't show up automatically.

brendongough, 2018-09-27:
I had the same issue:
https://github.com/vitotai/BrewPiLess/issues/101

vitotai, 2018-09-27:
That's weird. D1 mini pro is supposed to have 16M bytes flash instead of 4M bytes.
I have one which is used as usual D1 mini with 4M flash configuration. It works just fine. 

You can try modify the source like #101, or try modifying platformio.ini to have correct flash size:

find the configuration of your hardware setup, and change the line
`build_flags = -Wl,-Tesp8266.flash.4m.ld -DLegacyEspAsyncLibraries=true
`
change the 4m to 16m as

`build_flags = -Wl,-Tesp8266.flash.16m.ld -DLegacyEspAsyncLibraries=true
`