vitotai/BrewPiLess #128: Porting to ESP32

ferchinas, 2018-10-02:
Hi Vito:
There is so far some work done to port to ESP32?
I think there are many advantages as mensions in http://vito.tw/?p=720#comment-25109
-  TLS
-  More pins
-  BLE
-  And it is also less demanding on the source of energy.

I offer to collaborate to the best of my ability
Regards

allthepies, 2018-10-02:
Hi,

I did start a port attempt some time ago but my lack of C++ expertise made me stop, too many errors!

vitotai, 2018-10-04:
I tried my best to make it compiled. However, there seems like some library issue with platformio.
ESP32 uses whole different flash layout. I need to study it to make it work.
Before that, I am stuck on the LCD display with framework 1.8.0. 

You can try if you want.
https://github.com/vitotai/BrewPiLess/tree/ESP32Attempt

ferchinas, 2018-10-04:
Hello:
I tried to compile but as mentioned @allthepies, there are too many errors.

Have you been able to detect which libraries have conflict?
I am not familiar with the Brewpiless code, is it possible to isolate parts
of the code to be able to correct by parts?
Anyway, as soon as I have a time I will try the branch that you just shared
with me.

I will return with news,
Thank you

ferchinas, 2018-11-02:
Hi Vito:
I was with little time these weeks, yesterday I could compile and load it in my esp32.

This is the serial output:

Setup ()
SPIFFS.being () Success.
Init LCD ...
LCD Initialized ..
syscfg: 0, timeinfo: 156, gdc: 168, tempSchedule: 204, brewStatus: 300, logFileIndexes: 320, remoteLogginInfo: 636, autoCapSettings: 1080, parasiteTempControlSettings: 1092
Starting WiFi ...
AP SSID: BrewPi pass: BrewPi
AP Mode
WiFi Donate!
Load saved time: 36910
HTTP server started
{"" W "," SPIFFS space: 1347040
start volatiel log
mode ␀ => or
startVolatileLog, mode = o, beerteemp = 32767

Start as AP, I can connect to the network from my cell phone but I can not access 192.168.4.1

Do you have any suggestion?

zorce, 2019-07-24:
Also interested in an esp32 port. And will gladly help out.

exige81, 2019-07-28:
I also plan on digging into this with the ultimate goal of integrating the tilt hydrometer

vitotai, 2019-09-11:
Hi all, 
I made some progress. (Honestly, upgrading to 1.9 framework seems to be the biggest one.)
Please check here:
https://github.com/vitotai/BrewPiLess/tree/ESP32.test

The way to make ESP32 save network information seems to complicated for general user, or to me. Therefore, I coded to manually save it. There are some issues in the WiFiSetup Class, too. I found Auto Reconnect doesn't behavior as what I've thought.

The next step is defining PIN usage. What do you think?

allthepies, 2019-09-12:
Great stuff Vito!  What does it do so far ?

vitotai, 2019-09-13:
> @vitotai regarding AsyncTCP i assume you are getting the issue where the wifiscan data isn't being returned properly? At least that's where i've hit trouble in my own efforts. Scan results are being generated but the response is not being sent. Interestingly enough, every once in a while it works just fine.
> 
> Regarding pins, in my `config.h` i've avoided pins 12-15 as they are the hardware debug pins on the dev board i'm using (nodemcu-32s). Recently found the inexpensive ESP-prog JTAG debugger which works with PIO, and got it up and running using [this tutorial](https://www.youtube.com/watch?v=psMqilqlrRQ) perhaps that might be useful to those working on this.

Thank you for the head-up about PIN usages.  It is what I am looking for.
I too buy a JTAG debugger because I don't have any idea about the WString exception, which disappears after upgrading to 1.9 framework. I haven't setup the debugger, and thank you for the information, too.

If you turned on the debug message, you can see some "Too Many Message..." message, which is printed by WebSocket plug-in. My guts tells me that it might be the same cause of not-sending data and it might be something related to the "GreetingInMainLoop" option, which is a work-around for a design flaw I found in ESPAsyncTCP library. We'll see after I return home.

vitotai, 2019-09-17:
I was wrong. It's wasn't what I have thought, but I make it work by workaround without knowing the exact cause of issue.

The problem now is that the sensor cannot be detected.

ithinkido, 2019-09-19:
https://github.com/brettbeeson/esp32/tree/master/brew32

vitotai, 2019-09-20:
Some update:
1. DS18B20 doesn't work on certain PINs, like 25 and 26. I am not sure if it is just my board.
2. My latest push seems to work fine, although there are still some issues.
3. Working on Rotary Encoder now, but I need JTAG to work for an crash.

exige81, 2019-09-21:
Happy to see all of the positive progress on this, thanks so much for your effort.

I can report that PIN 25 and 26 are working with the DS18B20 both in BPL and simple test programs on my board. I'm using the NodeMCU-32S devkit.   Pins 25 & 26 are DAC pins but that really shouldn't matter so not 100% sure what would cause it not to work.

I've noticed a little bit of instability with the latest push, including a seemingly random crash/reboot, and the wifiscan is still giving problems, but at least everything seems to be generally working. I need to order some more hardware and may be able to help diagnose some issues in the coming weeks when I can find the time.

Thanks again

vitotai, 2019-09-21:
Basically, the rotary encoder works. However, I am stuck at a exception:

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

Mine works fine. At least it survived through 24 hours if I don't touch anything. Do you using "local logging"?  I am guessing something wrong with FLASH writing.

vitotai, 2019-09-22:
https://youtu.be/B2wUG_49lAk

The rotary encoder is working now! 

I went to wrong directions because of not familiar or ignoring Harvard architecture of ESP32.

vitotai, 2019-09-23:
The WiFi scanning issue should have something to do with "WiFi Auto Reconnect", which will causes disconnection in AP mode. It might because that WiFi library uses all resource to search lost network.
Try this branch. (My git tree are a mess now, and I can't push to previous branch.)
https://github.com/vitotai/BrewPiLess/tree/ESP32.trial

exige81, 2019-09-23:
Just uploaded that branch and I can confirm the WiFi issue seems to be resolved and is working properly.  Good job!   

Did notice that the latest doesn't seem to be detecting my onewire temp sensors, but that could be an issue on my end.  I'll look into it deeper once i get some more time.

vitotai, 2019-09-24:
Try the latest push, which includes an OneWire library.
The discussion:
https://github.com/espressif/arduino-esp32/issues/1335

I've tried back and forth, with and without this OneWire library. Sometimes it makes differences, sometimes it doesn't. It should depends on the run-time condition of I-CACHE.

exige81, 2019-09-24:
That seems to have solved it for me.  Thanks!

allthepies, 2019-09-25:
Compilation errors on ESP32.trial:

src/TempControl.cpp: In static member function 'static void TempControl::increaseEstimator(temperature*, temperature)':
src/TempControl.cpp:500:55: error: invalid operands of types '__gnu_cxx::__enable_if<true, double>::__type {aka double}' and 'int' to binary 'operator
>>'
  temperature factor = 614 + constrainTemp(abs(error)>>5, 0, 154); // 1.2 + 3.1% of error, limit between 1.2 and 1.5
                                                       ^
src/TempControl.cpp: In static member function 'static void TempControl::decreaseEstimator(temperature*, temperature)':
src/TempControl.cpp:510:55: error: invalid operands of types '__gnu_cxx::__enable_if<true, double>::__type {aka double}' and 'int' to binary 'operator
>>'

exige81, 2019-09-25:
@allthepies  This is an issue with the latest version of arduino-esp32 (v1.0.3), which is now used in the latest version of platform-espressif32 (1.10.0).    I've submitted the issue with espressif [here.](https://github.com/espressif/arduino-esp32/issues/3231)

For now you can fix the issue in your platformio.ini by setting `platform = espressif32@1.9.0`

allthepies, 2019-09-26:
Thanks, compiles with that workaround.

vitotai, 2019-09-26:
I've experienced compiling errors, unexpected exceptions, and etc. Upgrading to framework 1.9.0 seems to solve most of the problems.

Now that all functions I've thought of are working, I am thinking the next step.
BLE has been tried, but it takes too much memory to work with. I will wait until NimBLE library is available.
HTTPS seems not yet supported on ESP32. 
What do you have in mind?

exige81, 2019-09-27:
Eventually i'd like to get the Tilt Hydrometer working which will require BLE.  I'm sure you've seen the [tiltbridge](https://github.com/thorrak/tiltbridge) project.  Hydrometer data doesn't need to be checked continuously so maybe it would be possible to poll for data every 15 minutes so that might be a viable way to work around the memory issues?

Maybe a next step would be to abstract out the gravity code so that in the future it would be easier to switch devices since everything is tightly integrated around the iSpindle.

vitotai, 2019-09-27:
Frequency doesn't matter in this case. It's about memory, not computing power.

It's easy to integrate Tilt. The problem is that I don't have a Title to test.

Naesstrom, 2019-10-02:
Saw the vid that you got the potentiometer to work, great job!
Have you decided on a pinout yet? Was thinking of making a few PCB's for the ESP32 version!

stefschin, 2019-10-02:
Hi, I can help to make a PCB design for this project. Let me know if you're interested

vitotai, 2019-10-03:
> Saw the vid that you got the potentiometer to work, great job!
> Have you decided on a pinout yet? Was thinking of making a few PCB's for the ESP32 version!

It's a rotary encoder, not potentiometer.  They look alike, but different. BTW, the one I am using is KY-040 module.

> Hi, I can help to make a PCB design for this project. Let me know if you're interested

Any help is welcome. I don't have the ability to design PCBs, neither can I know how good or bad the design is.

So far, the PIN allocation is listed in the table.
https://github.com/vitotai/BrewPiLess/blob/ESP32.trial/doc/ESP32Pins.md

stefschin, 2019-10-03:
Ok, no problem, I will use [ESP32 Kit V1](https://www.aliexpress.com/item/32847027609.html), ok? 
Optional external Supply input, and which connectors you prefer? For sensors and buttons I like [Terminal Blocks](https://www.digikey.com/products/en?keywords=277-1721-ND). For display and Relay Board we can use same pins as modules, or did you prefer On board Relay?

vitotai, 2019-10-04:
I don't know the differences between ESP32 boards. I just picked what I can get from the vendor.

allthepies, 2019-10-04:
Regarding the PCB.  My idea scenario would be for a similar board to this one:

http://vito.tw/brewmaniacex-brewshield/

I use this board for my brewfridge controlled by BrewPiLess and a friend uses the same board to control his brewing kit using BrewManiacEx.  With ESP32 it could probably be simplified as the port expander may not be required.

Hwerow, 2020-01-03:
I have built the hardware to try out the ESP32 and rotary switch and have modified the IO in config.h #define rotarySwitchPin 25 to 27 as it was double allocated. I built it in VSC and tried to upload to the ESP32 but got a load of what looked like software error messages. 
Then I tried to upload the bin file v4.0a2.esp.bin using the ESP Download Tool V.3.6.8 and Win7 to see if that worked but with no success. I ended up with this - see screen shot - but I don't know what I am doing! Please could you help.
[image]

vitotai, 2020-01-05:
I never used  ESP Download Tool V.3.6.8 and can't help.

Hwerow, 2020-01-05:
OK . Please could you tell me what you did to upload and then how did you access BPL on the ESP32

vitotai, 2020-01-05:
I use PlatformIO with VSCode to upload.

Hwerow, 2020-01-05:
I have been using VSCode and PlatformIO but am getting errors with 4.0Pre when building for ESP32 like below

Processing esp32 (platform: espressif32@~1.9.0; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 2.10002.190628 (1.0.2)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.10.0
|-- <ESP8266_SSD1306> 4.0.0
|   |-- <Wire> 1.0.1
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.0
|-- <WiFi> 1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <AsyncTCP> 1.1.0
|   |-- <WiFi> 1.0
|   |-- <Time> 1.6
|   |-- <FS> 1.0
|   |-- <ArduinoJson> 6.10.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncTCP> 1.1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <WebServer> 1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <EEPROM> 1.0
|-- <Adafruit ADS1X15> 1.0.1
|   |-- <Wire> 1.0.1
|-- <Update> 1.0
|-- <FS> 1.0
|-- <OneWire> 2.3.2
|-- <Wire> 1.0.1
|-- <Time> 1.6
Building in release mode
Compiling .pio\build\esp32\src\Actuator.cpp.o
Compiling .pio\build\esp32\src\ActuatorArduinoPin.cpp.o
Compiling .pio\build\esp32\src\ArduinoFunctions.cpp.o
Compiling .pio\build\esp32\src\AutoCapControl.cpp.o
Compiling .pio\build\esp32\src\BPLSettings.cpp.o
Compiling .pio\build\esp32\src\BrewKeeper.cpp.o
src\BPLSettings.cpp:334:0: warning: "SECS_PER_MIN" redefined
 #define SECS_PER_MIN  (60UL)
 ^
In file included from C:/users/peter/.platformio/lib/Time_ID44/time.h:1:0,
[... 382 lines cut ...]
                 from .pio/libdeps/esp32/ArduinoJson_ID64/src/ArduinoJson.hpp:33,
                 from .pio/libdeps/esp32/ArduinoJson_ID64/src/ArduinoJson.h:9,
                 from src\BPLSettings.cpp:1:
C:/users/peter/.platformio/packages/framework-arduinoespressif32@2.10002.190628/cores/esp32/Print.h:86:25: note: forward declaration of 'struct tm'
     size_t print(struct tm * timeinfo, const char * format = NULL);
                         ^
*** [.pio\build\esp32\src\BPLSettings.cpp.o] Error 1
In file included from src\BrewPiProxy.h:5:0,
                 from src\BrewKeeper.cpp:5:
src\BrewKeeper.cpp: In member function 'float BrewProfile::tempByTimeGravity(time_t, Gravity)':
src\Config.h:514:53: warning: format '%lu' expects argument of type 'long unsigned int', but argument 7 has type 'uint32_t {aka unsigned int}' [-Wformat=]
 #define DBG_PRINTF(...) DebugPort.printf(__VA_ARGS__)
                                                     ^
src\BrewKeeper.cpp:250:2: note: in expansion of macro 'DBG_PRINTF'
  DBG_PRINTF("*tempByTimeGravity:now:%ld, step:%d, type=%c, last elapsed:%ld, step duration:%lu\n",time,
  ^
src\BrewPiLess.cpp: In static member function 'static void StatusLine::_printTime(time_t)':
src\BrewPiLess.cpp:1688:13: error: aggregate 'tm t' has incomplete type and cannot be defined
   struct tm t;
             ^
*** [.pio\build\esp32\src\BrewPiLess.cpp.o] Error 1
============================================ [FAILED] Took 5.79 seconds ============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

vitotai, 2020-01-05:
Something wrong with your platform, IMO. Manually Delete ".platformio" directory under the project directory and try again.

Hwerow, 2020-01-10:
After many attempts, I have got the ESP32 and rotary Encoder working. The encoder switch push button is on pin 27.
I did what you suggested but I could not get the 4.0pre to build and upload on my Win7 PC yet other sketches worked. So I used my Ubuntu laptop and it built, albeit with errors (see attached). I could only get it to upload by commenting out ; upload_port = /dev/cu.SLAB_USBtoUART in the platformio.ini file.

Seems to be functioning OK. Next step solder it up and put it into a box
Thanks for all your great work Cheers
PS I could not get the classic interface to work

[upload details.docx](https://github.com/vitotai/BrewPiLess/files/4043693/upload.details.docx)
[image]

vitotai, 2020-01-11:
The classic frontend was broken. It is temporarily fixed now. I am going to drop classic frontend at some time.

Naesstrom, 2020-02-11:
Quick question, finally got around to assemble all the parts for the esp32 version. Should I use the esp32 branch or the v4 one?

vitotai, 2020-02-12:
v4.0pre branch.
Maintaining multiple branches is a PITA.

Naesstrom, 2020-02-12:
> v4.0pre branch.
> Maintaining multiple branches is a PITA.

hehe yeah I know... hate it! I'll flash it and try it out then! Just starting with the rotary, lcd, 2xds18b20 and a double relay. Ordered a 5V pressure sensor that I'll connect later!
Was looking at the pins for the esp32 and there are a lot more actuators... what are the suggestions of use for them?
[image]

Naesstrom, 2020-02-13:
And as a PCB
### TOP layer
[image]
### BOTTOM layer
[image]

appi1, 2020-04-27:
is there a PCB for the ESP32 available?  I want to swap from brewpiless esp8266 to ESP32 Version.

exige81, 2020-04-28:
Concerning BLE, it looks like plaformio-espressif32 [released version 1.12](https://github.com/platformio/platform-espressif32/releases/tag/v1.12.0) which added support for ESP-IDF 4.0, so that means that nimBLE should now be available to use.   I haven't had time to try it out yet.

vitotai, 2020-04-28:
> > v4.0pre branch.
> > Maintaining multiple branches is a PITA.
> 
> hehe yeah I know... hate it! I'll flash it and try it out then! Just starting with the rotary, lcd, 2xds18b20 and a double relay. Ordered a 5V pressure sensor that I'll connect later!
> 
> Checking the pinout I see that the 2 remaining actuators can be used for fan & door but can't find the functions described anywhere. The fan is for circulation and door to?
> 
> [image]

Sorry to miss your question. As far as I know, the door sensor is used to activate light, if assigned, and add annotations to chart. However, BrewPiLess doesn't show it on the chart, the display of LCD should show, though.

vitotai, 2020-04-28:
> is there a PCB for the ESP32 available? I want to swap from brewpiless esp8266 to ESP32 Version.

I don't know. A friend sent me one for reviewing, but I don't have the ability to review.

stefschin, 2020-04-28:
> 
> 
> is there a PCB for the ESP32 available? I want to swap from brewpiless esp8266 to ESP32 Version.

I'm on it...hope today I finish it and send to PCBWay if somebody wants to build it. Did you have ESP32-DOIT-DEVKIT-V1-Board  with 30 or 36 GPIOs?

appi1, 2020-04-28:
great news,  my Board is the following  with 38 pins https://de.aliexpress.com/item/4000134023059.html?spm=a2g0s.9042311.0.0.27424c4diK4lbw

stefschin, 2020-04-28:
I've found 3 models, with 30, 36 and 38 pins and they are not compatible to make 1 board. I can make 1 from each other if @vitotai want. And you prefer SMD or through hole components?

appi1, 2020-04-28:
the version with 38 pins is ok for me, thank you very much, SMD is ok. is it possible to integrate a DC Converter as AMS1117, so we can use 12V for the PCB and also for the relays.

lalo-uy, 2020-04-28:
Be carefully with AMS1117, to input is 12v . With 12.5 or 13 burn in 12v
pass thru.
4 esp8266 smoked that way.
Old good 7805 or 78M05 mutch better.

appi1, 2020-05-08:
hi Guys
some news to the PCB?

stefschin, 2020-05-11:
Hi! Give me some days to finish it, I was very busy these days

fabiojavaroni, 2020-06-28:
Hello Guys,
I finally got to assemble all the parts for the esp32 version.
I´m using a Oled + rotary + esp32 + 2 channel relay + 2 ds18b20.

Using the tips in this posta i could upload with Vscode and Platformio.
However the oled isnt working and i don´t know the reason.

Have i need to turn on something in frontend?

Environment            Status    Duration
---------------------  --------  ------------
esp32-dev              IGNORED
esp32                  SUCCESS   00:00:52.666
esp32-oled             IGNORED
esp8266dev             IGNORED
classic                FAILED    00:00:45.795
d1pro-classic          FAILED    00:00:43.905
d1pro                  FAILED    00:00:45.761
classic-spanish        FAILED    00:00:37.376
classic-portuguese-br  FAILED    00:00:44.170
classic-slovak         FAILED    00:00:37.015
classic-tw             FAILED    00:00:24.978
classic-it             FAILED    00:00:43.910
newui                  FAILED    00:00:19.642
newui-spanish          FAILED    00:00:45.735
newui-portuguese-br    FAILED    00:00:43.172
newui-slovak           FAILED    00:00:42.643
newui-tw               FAILED    00:00:42.355
newui-it               FAILED    00:00:42.869
newui-oled             FAILED    00:00:44.383
thorrax                FAILED    00:00:20.144
thorrax-classic        FAILED    00:00:48.503
sonoff                 FAILED    00:00:44.213
sonoff-esp8285         FAILED    00:00:44.012
========================================= 19 failed, 1 succeeded in 00:13:33.249 ========================================= 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Thank you

allthepies, 2020-06-29:
> esp32-oled IGNORED

That ^^ looks to be the reason, you haven't built the OLED version for ESP32

fabiojavaroni, 2020-06-29:
How i do this?
Thanky you

sedgington, 2020-07-01:
OK, I'm responding to this post to help anyone else out there that has not gone through the (initial) pain of trying to get the ESP32 chip to work with brewpiless. While I have never had a problem with the 8266 D1 mini and the excellenbt .bin files of brewpiless, the ESP32 would never work for me with any of the .bin files I tried--either from the -4.0pre branch or from the Master branch.  While Nodemcu-PyFlasher-4.0 on my macbook pro repeatedly would successfully load the .bin file to my ESP32, I would never get the brewpiless AP to come up so that I could then log in. At first, I thought it must be something wrong with the development board I had ordered: 
https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1 
so I ordered a different board: 
https://www.amazon.com/gp/product/B07DBNHJW2/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
same story.
This sent me down the road of building and uploading my own .bin file using Visual Studio Code with PlatformIO--again on the same macbook pro. I'm happy to say this was successful in the end, but unlike Nodemcu, I had to physically press the boot or Ioo button and then (sometimes) the reset button while still holding down the boot button to get the board to go into bootloader mode so that PlatformIO could communicate with the device ( the last board of the second group of boards, balked on communicating through the USB port if I hit the reset button and would only work when I hit the boot button). 
So here are is the short version of the steps I had to do to get the ESP32 boards to successfully load brewpiless: 
1. Download Visual Studio Code and install it
2. Add PlatformIO to VSC
3. Download the brewpiless-4.0pre branch from github: https://github.com/vitotai/BrewPiLess/tree/v4.0pre
4. Unzip this branch and copy it to a subdirectory of PlatformIO (mine was in the Documents folder)
5. Open this folder that contains the platformio.ini from within VSC/Platformio
6. Click on platformio.ini so that it is open in PlatformIO. 
7. At the top of the file find the line default_envs = esp8266dev under [platformio] and change it to #default_envs = esp8266dev
8 Under that line add: default_envs = esp32-dev
9. Save the file
10. find the line  
upload_port = /dev/cu.SLAB_USBtoUART under [env:esp32-dev]
and change it to upload_port = /dev/cu.usbserial-0001 (I knew this was my usb port because Nodemcu had told me that information when I went to load the .bin file).
11. Do the same for the line under it that begins monitor_port =
12.  Under "View" in VS Code, click on "Command Palette" begin typing in PlatformIO: Build until it finds the command. It will then run the build command.
13. Connect your development board to the USB port of your computer. Be sure to use a cable that is capable of transmitting data and is not just a power cable. The red led will come on. Either push and hold the "boot" or "Ioo" button and while still holding it down, briefly hit the "reset" button, then release the button you have been holding down. (if this doesn't work, just hold down the "boot" button and release it.)
14. Assuming that the "PlatformIO: Build" command ran successfully, repeat the steps of 12. but begin typing in "PlatformIO: Upload" until the command is in the box. It will then run automatically.
15. If you successfully found the correct upload port for your usb connection to the board, the program will run and at the bottom of the console you will see under the line that begins with "Environment" the line:
esp32-dev              SUCCESS    00:00:43.070
All the other Environments will state "IGNORED".
16. Unplug the development board from the USB cable and then plug it back in.
17. Monitor your wireless network for "brewpiless" and click on that network. Open a web browser after you successfully join brewpiless and type in 192.168.4.1 to bring you to the homepage of brewpiless. Go to the system tab, scan for your local network, select it, put in the password, save it and then, importanly, hit the "submit" button on the bottom right part of the screen. Brewpiless will join your network and you can set it up as desired.

Now, why is this pain worth it? Well, the ESP32 seems to be much faster, has more memory, and should be more stable than the 8266 boards. In addition, it has Bluetooth built in which means that down the road it might support Tilt Hydrometers as well as the currently supported wifi-enabled iSpindel Hydrometers (which are difficult to source in the USA because Tilt has apparently intimidated the German supplier of iSpindel kits to the degree that he will not ship to the USA).

I hope this helps someone and saves them from the pain of getting their ESP32 board to work. If I have missed something or there is a better way, please comment on this post on your way of doing this.

sedgington, 2020-07-02:
As an addenda to my post above, I found that my driver problem listed in step 9 above:
upload_port = /dev/cu.SLAB_USBtoUART under [env:esp32-dev]
is apparently a common problem in the Mac operating system and to a degree with Windows as well. About the only OS that seems to not have encountered this issue reguarly is Linux. However, this is easily fixable (at least for me). If you navigate to the URL below and dowload the drivers for your operating system, you can then install them:
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
Once installed, I tested them within visual studio with the platformio extension by downloading the recently updated Master branch and then following my steps above without changing the upload port. A word of caution for the noviate: this version will build all the bin files of the branch unless you only specify the environment that you want at the top of platformio.ini. Once specified, I successfully built and loaded my ESP32 with the new drivers. However, Nodemcu would still not load a working copy of brewpiless from the precompiled .bin file in the master branch--even though it said it had successfully loaded the .bin. In any case, one small request for the various versions: Is there a place within brewpiless that one could see exactly which version has been loaded. Mine still says v4.0 after uploading the new Brewpiless.4.0r2.trial.esp32. Thanks!

vitotai, 2020-07-08:
@sedgington 
I would like to add your instruction to the document, if you don't mind.

[13 more replies]