vitotai/BrewPiLess #130: Secondary Web Interface (How To)

dotps1, 2018-10-05:
I read (at some point) there was a secondary refactored web interface that someone had created, is there any how to's on how to set it up?  I read it was committed to this repo (I think).  thanks!

vitotai, 2018-10-05:
Download 
index.htm.gz, config.htm.gz, gravity.htm.gz, logging.htm.gz, setup.htm.gz, control.htm.gz
from the following folder:
https://github.com/vitotai/BrewPiLess/tree/master/htmljs/dist

open filemanager
http://brewpiless.local:8008/filemanager

upload the downloaded files. Don't forget to "reload" the pages.

dotps1, 2018-10-05:
extract them first?

vitotai, 2018-10-05:
No extraction. Use exact what they are named.

dotps1, 2018-10-05:
worked good, thanks.

closed 2018-10-05 by dotps1

vitotai, 2018-10-12:
yes. You can delete them, and the embedded files will be used.

dotps1, 2018-10-12:
forgive me, but I did not see a delete option in the file manager page.  only upload and create.

allthepies, 2018-10-12:
Right click on the file you want to delete, you should then get a popup menu with a delete option.

dotps1, 2018-10-12:
I feel like I tried that as well, however I will attempt it again when I get home.  Thanks!

dotps1, 2018-10-13:
yeah, right click doesn't do anything, using Firefox.