vitotai/BrewPiLess #132: Onewire temperature  sensor issue

FluxFace, 2018-10-09:
I have build an BPL and having problem with the temperature sensors.
When i connect one DS18B20 onewire sensor to BPL it shows up in BPL and works fine.
If i connect two DS18B20 BPL won´t find the sensors. What am i doing wrong her?

allthepies, 2018-10-09:
How are you connecting them ?  Bus style (as per the diagram link) ?
Do you have a 4K7 resistor between the data line and VCC ?
What voltage is VCC ?

https://cdn.instructables.com/F79/RUWE/GTUY9ZXQ/F79RUWEGTUY9ZXQ.LARGE.jpg

FluxFace, 2018-10-09:
Connecting them like the diagram you linked to. Yes i have a 4K7 resistor between the data line and VCC. The voltage is 3.3v I have tried with 5v  The strange is that it works with one sensor but not two.
I have tried to flash with V3 and upload with platformIO

FluxFace, 2018-10-09:
I have found the problem. It was a bad resistor.
Thank you for the help.

closed 2018-10-09 by FluxFace