vitotai/BrewPiLess #133: Error after update v 3.1

jaimeagaviria, 2018-10-10:
I uptaded to V3.1, when I try to open brewpi.local to setup wifi parameters then I get a modal dialog "Communicating with BrewPiLess controller", but the dialog never is closed, then I open the console in google chrome to check.

Failed to load resource
Dygraph is not defined

Maybe the web page is trying to connect external web service to get cdnjs.cloudflare.com.graph-combined.js

But  the nodemcu not is connected to internet because no have the wifi parameters.

I executed the next link too
http://brewpiless.local/erasewifisetting

but I dont get a webpage to setup the wifi parameters.

Thank you,

vitotai, 2018-10-11:
I see. Will try to fix ASAP.
Please go to "config.htm" directly to configure the network.

jaimeagaviria, 2018-10-11:
Thank you for your help, yes I opened config.thm page but in this page not see the option to configure the wifi connection.

Maybe is the page http://192.168.4.1/wifi?

I wiil try again because I Instelled version 3.0 again.

Now, I need change some parameters to PID; I notice that the cooling / heating cycles are long and this causes the beer to heat up or cool too much

Kind regards

vitotai, 2018-10-11:
You might need to "reload" the page.