vitotai/BrewPiLess #137: Wifi connection randomly(?) lost

flo269, 2018-10-21:
Hi,

I'm running v3.2 and during my current fermentation I am experiencing wifi connectivity issues. Brewpiless seems to lose connection to wifi: This happened twice in the last few days (see flat filtered SG graphs). I then pulled the plug to reset and everything worked fine again.

What could I do to deliver more precise info?

Cheers
Flo

[image]

flo269, 2018-10-21:
@amlynarczyk do you mean the bin file with the new ui or when you install the new ui manually?

I'm not sure if its a ui issue because I can't reach the ui in first place and I can't reach BPL via ping either.

amlynarczyk, 2018-10-21:
Bin file

vitotai, 2018-10-24:
I had the same issue. I've followed this issue for a while without luck.

The new ui version uses framework 1.8.0 with latest ESPAsyncWebServer library, while the classic one uses 1.5.0 with older version of ESPAsyncWebServer library.

I've tried classic ui with 1.8.0, but it was more unstable.

One possible issue I've found is trying to reconnect if BPL found it disconnected. I am trying to change it by letting the WiFi library do the job instead of doing it manually. 
The code will be available soon if the test platform doesn't crash for at least 24 hours.

flo269, 2018-10-25:
Awesome, thanks for the update @vitotai !

brendongough, 2018-12-02:
I found using 1.5 framework and v3.2 ended up with the wifi crashing, the device still held a constant temp which was nice. Since moving to v3.4 which has native 1.8 framework, problems solved :)

closed 2023-03-21 by flo269