vitotai/BrewPiLess #138: problem with version 3.2 on wemos d1 mini pro

sebpa, 2018-10-23:
I have a problems on wemos d1 mini pro. At first it didn't want to connect to my wifi. I scanned for networks, chosen my wifi, filled pass, and than after saving, only the name of my wifi appeared. IP was 0.0.0.0

I decided to develop on this problem. When I flashed with version 2.6 with old wifi manager the problem disappeared, board could connect to the wifi.

Then I realized that also .bin with 3.1 newgui version is working  (but on the other hand regular bin version of 3.1 wasn't - the same issue as above)

At last I configured new device in platformio.ini with this 

`[env:d1_mini]
platform = espressif8266@~1.8
board = d1_mini
framework = arduino
lib_extra_dirs = ./lib_async
build_flags = -Wl,-Tesp8266.flash.4m2m.ld
    -DUseNewFrontEnd=true`

after building and uploading it was ok, no issues at all, but it was only when I checkout 3.1 version and put this conf, 
the same config with the newest commit has different problem - when i connect to the board and try to visit http://192.168.4.1/ i got timeout.

vitotai, 2018-10-24:
Some D1 min pro seem to have issues.
Check this issue.
https://github.com/vitotai/BrewPiLess/issues/101

brendongough, 2018-11-10:
1.8 does crash frequently though. I just set wifi data static, however, I still find the web server crashes after ~72hrs. I will be changing all mine to genuine non-pro versions to avoid any odd issues.

brendongough, 2018-12-02:
Just an update on this, v3.4 works perfectly, no crashes and it's 1.8 native :)

terragady, 2018-12-07:
> 
> 
> Just an update on this, v3.4 works perfectly, no crashes and it's 1.8 native :)

Do you have better range using Pro version? Do you use ceramic antenna or external one?

brendongough, 2018-12-07:
I use iSpindels so the wifi router is within 1m of the devices, I probably don't "need" to use the Pro version however this is what I ordered for the iSpindels. You will get better reception using the Pro version, however, you do need to complete one SMD change, moving the bridge (0) from connecting the ceramic antenna to connecting the external IPEX antenna.

[image]