vitotai/BrewPiLess #142: Cant get the SoftAP page via browser

jimhiggs, 2018-11-11:
First let me say I am new to BPL and esp8266. I have used nodemcu flasher to flash my device with 3.3.
 I spent many hours just trying to find out how to connect to the config page without wifi access via my router. It would be helpful if the wiki page could explain just  a little more about how to access. 
So using an android tablet I worked out I needed to connect to Brewpiless ap. 
Which I did. I opened a browser and spent another couple of hours trying to work out what to put in the address bar. I tried 192.168.4.1 and /config and various port numbers but got unable to connect. I then tried flashing 2.6 and that works fine, I got into wifi manager and successfully configured my wifi. I then went back to the 3.3 version and just get unable to connect.

What more info would be useful?

jimhiggs, 2018-11-11:
tried 3.1 again and worked on third try. Will stick with this for now.

jimhiggs, 2018-11-16:
3.1 does not work either. Thought it might be the fact that i got my esp2866 from aliexpress so ordered 2 from amazon. No luck. Only version I can connect to via the router is 2.6. Would I likely have more success using platformio (i think that is what it was called)? Or maybe a different baud rate on the nodemcu flasher ? I leave it at default.

lalo-uy, 2019-03-11:
I have more or less the same problem. 2.6 shows up the Ap mode when no know WiFi is available.
After ota upgrade to 3.4-thorak it work OK with the configured network, but if moved away no AP comes up. Even as it was configured as Station+AP . Allways using pre compiled bins