vitotai/BrewPiLess #143: why v3.3r1newui does not allow me to access via 192.168.4.1 to complete config

Hwerow, 2018-11-13:
I think BPL is great and I have been using V2.7 successfully for a number of brews and decided to upgrade to the new ui while keeping the oled functionality. I have been using the compiled bins.
I tried v3.1 newui which allows me to set my fixed IP so I am able to access BPL from 192.168.4.1 AP and my fixed IP address - all good
But using 3.2r2 newui or v3.3r1newui does not allow me to access via 192.168.4.1 to complete config.
So I wondered what was changed between v 3.1 and 3.2 that might be causing me this problem?
I also tried the versions 3.1, 3.2 and 3.3 oled with no success. I assumed that the oled versions would be a variation to the newui but perhaps not?

Hopefully, this is something simple that can be fixed
Cheers
Peter

vitotai, 2018-11-13:
Starting from 3.1, WiFiManager is no longer used.

"newui" options use framework 1.8.0 while all other options kept using 1.5.0 until 3.2. 

After 3.2, all options use 1.8.0 frameworks. ESPAsyncWebServer and ESPAsyncTCP libraries are also updated to keep with latest framework.

Please note: OTA update "newui" variant to/from non newui variants will cause unpredictable problem due to the different flash  layout.

Hwerow, 2018-11-13:
OK, for me 3.1 bin works in terms of being able to configure but 3.2 bin  does not allow me to configure
Is there something broken in 3.2?

vitotai, 2018-11-13:
Yes. 3.2 and 3.3 has issues saving configuration. (I was in the middle of developing new features that broke the configuration saving function.)
3.3r1 fixed it.

LucLN, 2018-11-13:
for me the same ..

vitotai, 2018-11-13:
Try RELOAD the configuration page.

LucLN, 2018-11-13:
> I can't get to the config page as I can with v3.1
> Using v3.3r1newui - I have spent a couple of hours with this version. I can see the brewpiless signal but cannot login using brewpiless/brewpiless 192.168.4.1 have tried 3 different OS iPad, Linux and Win 7 with no success.
> Suggestions appreciated

for me the same ..

vitotai, 2018-11-14:
https://github.com/vitotai/BrewPiLess/wiki/Flash-Layout-and-New-Frontend

Hwerow, 2018-11-14:
I'm sorry but while I can read the words in your link, I don't know what I should do to resolve my problem.
Is there an erase and start again function available? 
Do I need to use a new 8266 to install 3.3newui? 
I am looking for a solution.
Thanks

amlynarczyk, 2018-11-14:
You need to connect your board to a computer and flash it using usb cable, not over the air update.

Hwerow, 2018-11-14:
I have used a USB cable and the nodemcu windows flasher to upload 3.3r1newui

vitotai, 2018-11-15:
Can you access the first page, index.htm? 
If so, try formatting the SPIFFS by using the URL:
http://brewpiless.local:8008/format-spiffs

Under SoftAP mode, It will be 
http://192.168.4.1:8008/format-spiffs

LucLN, 2018-11-15:
.. for me the same .. tried it on nodemcu and d1 mini, both used and fresh devices ..

Hwerow, 2018-11-15:
Can you access the first page, index.htm? No, below is what I tried - 
I loaded 3.1newui, used the format spiffs, as I could get to index then loaded 3.3r1newui - I could see the brewpiless AP but could not access using brewpiless as the password so I could not get to the index page

Then I swapped out the 8266 with a different one and tried to load 3.3r1newui same problem. Loaded 3.1newui and it lets me do the configuration.

LucLN, 2018-11-15:
same for me .. just tried again ..

vitotai, 2018-11-15:
it's weird.

The modification from 3.1 to 3.2 is mostly at html files and framework version, and 3.2 to 3.3 is replacing ServerSideEvent to WebSocket.

If your nodemcu/d1 mini had connected to your network at 3.1, it should connect to the same network after upgrading to 3.3r1.

3.2 and 3.3 DO have a bug that makes saving configuration impossible. That should not affect the network connectivity.

Hwerow, 2018-11-17:
I have just uploaded V3.4 and am able to access via both 192.168.4.1. and my fixed IP addresses.
Thanks whatever you did has resolved my issue Peter

terragady, 2018-12-07:
> "newui" options use framework 1.8.0 while all other options kept using 1.5.0 until 3.2.
> 
> After 3.2, all options use 1.8.0 frameworks. ESPAsyncWebServer and ESPAsyncTCP libraries are also updated to keep with latest framework.

so even old UI uses the new framework now? So the difference is only the frontend look right?