vitotai/BrewPiLess #144: POST Remote Logging not working to BrewFather.app with bpl v3.3

dotps1, 2018-11-15:
After updating to bpl 3.3 the POST remote logging to the brewfather app is no longer working.  I have reached out to brewfather and after verifying my configuration they said I will have to downgrade if its not working.  So I thought I would ask here if there is anything I can provide to help figure this out.

my wifi seems fine, I have not had it drop at all when access the web interface.

thanks.

vitotai, 2018-11-15:
What is your old version that you upgraded from?
I just tested my setup, and it works.

dotps1, 2018-11-15:
it must have been pre v3, because I had to re apply all my settings.

vitotai, 2018-11-16:
Based on all information I have, I would suggest to make sure you have correct settings or downgrade to what works for you.

dotps1, 2018-12-18:
Just to let you know, I just updated to v3.4r1 and its working again. I really like the brewfather 'service' type.  One thing I hope you guys can get working is allowing me to have F on my controller and have that mess up the logging in the brewfather app.  thanks!

closed 2018-12-18 by dotps1