vitotai/BrewPiLess #147: my update to V3.4

LucLN, 2018-11-17:
Hi vitotai,

just updated my devices to 3.4.oled and everything, I mean the connection to AP & local wifi, work fine .. thanks for that !

Do you have any plans for compiling a .bin with newui & oled ?

Greetings from Luxemburg,

Luc.

[image]

vitotai, 2018-11-18:
glad that it works for you.
My browsers cached the files, so I didn't find the issue until I tried to force reloading.

It can be done easily, but there are too many combinations of options, including language, frontend, LCD, and etc. I must consolidate them.

LucLN, 2018-11-20:
Hi, 

> It can be done easily, but there are too many combinations of options, including language, frontend, LCD, and etc. I must consolidate them.

so you want to motivate me to setup my own development environment and compile it :-)

brendongough, 2018-12-02:
I compile and install mine with Visual Studio and Platform.io - It took maybe an hour or so to get it set up and my head around it, after that it's super easy.

Hwerow, 2018-12-02:
While I have VS and Platform IO I have compiled versions before - But don't know how to select and compile the relevant stuff/files to make a 3.4 oled newui version Some guidance would be most useful Thanks

brendongough, 2018-12-07:
What happens when you try to compile?

There is a comment in 3.4 saying newui is not yet working. I'm unsure if there's more context around this however I assumed it wouldn't work.