vitotai/BrewPiLess #148: Adding support for rf 433mhz transmitters

Hammersoe, 2018-11-21:
A feature that would be great to see combined with BPL is RF remote controlled power sockets. Instead of using relays (or in addition), a simple rf transmitter could engage or disengage heater or cooler using very common remote controlled sockets, which btw are quite cheap.

vitotai, 2018-12-10:
I am able to program a little but not able to create hardware. If it is clear about how to control, I might be able to program it given the fact that the hardware can do it.

Lai1337, 2019-01-02:
Support for 433 mhz sockets would be highly appreciated. Support would enable usage of BPI without the need to handle AC installations.

From my own viewpoint, I am trying to make HomeEasy (self-learning sockets, such as Nexa) to work.
http://playground.arduino.cc/Code/HomeEasy
https://github.com/nbogojevic/piHomeEasy