vitotai/BrewPiLess #149: slovak locale

petobena, 2018-12-05:
Hi Vito,

I created slovak locale, can we copy my files to your repo?

few strings must be still translated, but I have it almost done

petobena, 2018-12-12:
what is the meaning of "Cap" "Capping" "Capping Status" etc. ?

lalo-uy, 2018-12-12:
Closing the air lock in the last few days of fermentation for natural
carbonating the beer.

petobena, 2018-12-13:
can you take all files you need from here?
https://github.com/petobena/BPL

petobena, 2019-02-03:
@vitotai 
hi Vito,

here is my slovak update for the new version: https://github.com/petobena/BrewPiLess/blob/master/htmljs/src/locales/slovak.json

and I have some bugs, should I write a bug?
do you need some help with test? I work as software tester so maybe I can help