vitotai/BrewPiLess #151: Issue with Ispindel

Orion5512, 2018-12-15:
Hi 
I installed firmware 3.4 in my BPL D1 mini, it works fine but does not connect with Ispindel.
i have setting for Ispindel support but the Ispindel window does not appear.
In the Ide Arduino serial monitor when Ispindle try to connecet with BPL there is this log
17:08:02.948 -> FW 6.0.0
17:08:02.948 -> 2.2.1(cfd48f3)
17:08:02.948 -> Worker run!
17:08:02.948 -> mounting FS... mounted!
17:08:02.985 -> reading config file17:08:02.985 -> applying offsets
17:08:02.985 -> parsed config:
17:08:02.985 -> {"Name":"ispindel1","Token":"","Sleep":900,"Server":"192.168.4.1","API":3,"Port":80,"URL":"/gravity","DB":"ispindel","Username":

"brewpiless","Password":"brewpiless","Job":"ispindel","Instance":"000","Vfact":200,"TS":0,"OWpin":12,"SSID":"brewpiless","PSK":"

brewpiless","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":-1217,"aY":-757,"aZ":965}
Boot-Mode: Deep-Sleep Wake 
17:08:03.155 -> woken from deepsleep, normal mode
Samples:42 min:86.31 max:86.38 time:758
17:08:03.937 -> x: -1124 y: 17218 z: -106
17:08:03.937 -> Tilt: 86.34
17:08:03.937 -> Tacc: 21.06
17:08:03.937 -> Volt: 3.73
Temp: 19.44
17:08:03.971 -> Gravity: 31.73
17:08:03.971 -> IP: 192.168.4.317:08:03.971 -> 
17:08:03.971 -> calling HTTP
17:08:03.971 -> 

{"name":"ispindel1","ID":3309802,"angle":86.33989,"temperature":19.4375,"temp_units":"C","battery":3.73,"gravity":31.7264,"inter
val":900,"RSSI":-32}HTTPAPI: posting
code: 500
Can you help me please?

terragady, 2018-12-16:
did you set AP+Station mode in the BPL settings? You also need to start logging to see the gravity, at least it was a case in mine.

Orion5512, 2018-12-17:
Hi 
BPL firmware is **BrewPiLess.v3.4r1.oled.bin**
BPL is setting to Station+AP, logging is started, Brewfater receive data from BPL  without Ispindel data

[image]
[image]

Arduino Ide serial monitor log

[image]

19:46:52.429 -> IP: 192.168.4.2
19:46:52.429 -> 
19:46:52.429 -> calling HTTP
19:46:52.429 ->{"name":"ispindel1","ID":3309802,"angle":86.70926,"temperature":19,"temp_units":"C","battery":3.71,"gravity":31.91232,"interval":900,"RSSI":-30}HTTPAPI: posting
**code: 500** 
HTTP code 500 say Internal Server Error

I can not understand the cause of the Issue

terragady, 2018-12-17:
IP should be 192.168.4.1

Orion5512, 2018-12-17:
192.168.4.2 is Ip that  BPL Station assign to Ispindel  when it connect to.

vitotai, 2018-12-23:
Please change the name of iSpindel to starts with "iSpindel", like iSpindel11. 
The lower case "ispindel" is not recognized.

Orion5512, 2018-12-30:
Thanks for the suggestion, iSpindel work