vitotai/BrewPiLess #152: Support F temp when logging to BrewFather

dotps1, 2018-12-18:
is there anyway to support Fahrenheit temps when sending the data to brewfather.app?

[image]

basically, because I have Fahrenheit set in brewfather, it just upconverting the Fahrenheit values sent from the brewpiless as if the where Celsius.

thanks!

vitotai, 2018-12-23:
It seems like brewfather assumes that the temperature values are always in Celsius. BPL just reports the value whatever unit it is.
Let me discuss with Brewfather for this issue.

dotps1, 2018-12-23:
thanks!  I think you will need some way to signal from bpl what unit it is.  if I have it in C on all systems its fine, but if I have my Brewfather preference set to F, and then have my bpl set to F, it goes F > C > F, so its being sent from bpl as F, then brewfather thinks its C, then upconverts the temp to F again, F > C > F.  hope that makes sense lol.

tgangso, 2018-12-23:
In version of brewfather 1.4.0 released 21.12.2018. I added a new field to the brewpiless logging format called "tempUnit".

tempUnit can be C for celcius, F for fahrenheit or K for kelvin.

It would be nice if you can add a %variable for that @vitotai and handle it automatically if the user selects Brewfather as logging method.

vitotai, 2018-12-25:
It will be available in next release., v3.5.
However, the setting must be updated again.

dotps1, 2019-02-18:
I see 3.5 is released, ill give this a try tonight.

dotps1, 2019-02-28:
I did try this, but im getting a "not enough free space" error when trying to update the BPL.

dotps1, 2019-03-03:
I got 3.5 on my BPL, but its still doing the conversion it appears:

[image]

dotps1, 2019-03-03:
there is also no where on the settings page to put the format string in anymore:

[image]

dotps1, 2019-03-03:
sorry to keep spamming this, but now its 'half way' working.  the actual temps are in F, but the set temps are not:

[image]

Hammersoe, 2019-03-09:
Had a quick look at this with the serial output enabled on the BPL. It looks like the BPL sends the units correctly if you have changed it to F instead og C, so the conversion issue might be on the Brewfather end.

Output from a test unit shows that both actual and set temps are sent in F to Brewfather: "{"id":"BPLTEST","beerTemp":76.9,"beerSet":68.0,"fridgeTemp":77.0,"fridgeSet":50.0,"roomTemp":null,"gravity":null,"tiltValue":0.00,"auxTemp":null,"extVolt":null,"timestamp":0,"tempUnit":"F","pressure":null,"mode":"b"}"

@tgangso Are you aware of this still being an issue?
If the units are set to F on BPL, and C on Brewfather the conversation to C only happens on the Temp and not on the Set Temp. If both are set to F then the Temp is reported correctly, but the Set Temp seems to be converted from F to double F.

tgangso, 2019-03-09:
Thanks for letting me know, I did check my code and you are correct, I am not doing the conversion for the set temps, will fix!

dotps1, 2019-03-15:
I am closing this, I have verified that all data is reading correctly.  Thanks everyone!  I really appreciate everyones help.  and keep up the good work!

closed 2019-03-15 by dotps1