vitotai/BrewPiLess #153: Change AP IP

jimhiggs, 2018-12-18:
I have managed to get 3.4r1 working on two of my  4 esp's but can only get them to run in AP only mode.
I want to run two, one for ferm chamber and one for Glycol chiller. I want to change the AP IP of one of them so they can both be accessed in the same room (unless there is a better/other way) . 
I am now using Platformio and building the release myself but have searched all files in the rep and can't locate where the IP is configured?

vitotai, 2018-12-23:
I don't understand why you can run in AP mode only.

You don't need to change the default AP IP, but you might need to change the "host/network name". Otherwise, I don't know how you can differentiate them.

vitotai, 2018-12-25:
When you connect one of the BrewPiLess in AP mode, 192.168.4.1 is the BPL you connect. You can't connect TWO networks at the same time (well, in most case.), so you cannot connect to TWO BPL in AP mode at the same time.
I don't really understand your issue. You should try to configure BPL to connect to your router.

jimhiggs, 2018-12-26:
Thanks for the reply Vitotai.  I have configured the BPL to the same IP as my router but it does not show up in my routers DHCP list and even with a fixed IP, I cannot connect to it from the same network.

I realise that two devices trying to host AP's on 192.168.4.1 will not work, that is why I was asking if I can change one of them to use say 192.168.5.1. Then I will be able to connect to either with my tablet.

jimhiggs, 2019-01-06:
I tried again to get access via router today and in system config, I am having a lot of trouble getting it to save changes to the Network (SSID) and fixed IP. I cant get it to clear the IP at all. It clears but is not saved so next time I go in its still at the fixed IP. However, instead of accessing the AP from my android tablet and phone, I tried accessing from a wireless W10 laptop and it seemed easier to get the changes to stick,  I still can't change the IP, but I changed the SSID  to empty and then after I got it to save that (took quite a few attempts), hooray,  I could finally access via the fixed IP..... So I will leave it at that for now. 

Although I would still like to be able to change one of the AP IPs.  
So can you tell me where the IP is set at 192.168.4.1 in code  so that I can change it?

jimhiggs, 2019-01-06:
Perhaps it would be a useful feature to be able to configure the AP Subnet.  I for one will have more than one BPL running in an area with no Wifi available.
Is it possible to configure a second BPL to be Station only and set its Station IP to be a different IP on 192.168.4  and it then be  accessible via the first BPL? Or is that not how it works?

quantum8, 2019-01-07:
I think there may be some confusion with terminology here!

If you're running the device as an AP, it means that the device will make it's own wifi network in which you connect to in order to view and/or configure it. Putting the device into station mode will allow the device to connect to your router and request it's IP address from the router via DHCP. This is when you'll be able to see it listed in the router's list.

It sounds like you have two devices, each running it's own wifi access point. Connecting to one device using 192.168.4.1 does not interfere with the other device as it is on its own separate network. There is no bridge or route between the two networks. This is why you can have two devices with the same IP addresses as they cannot see each other.

If you switch both of the devices on, you should be able to see two separate entries for the "Brewpiless" network. You should probably change one of them to use a different network name if you want to differentiate between the two.

I had a load of issues trying to get mine to connect to my router using the interface so I forced it to from the code. Within WiFiSetup.cpp, there will be a section that sets up the wifi access to a router, WiFiSetupClass::begin. 
Look for the WiFi.begin() line and insert your router's wifi credentials so it looks like this, including the quotes: WiFi.begin("YOUR_SSID", "YOUR_WIFI_PASSWORD");

Recompile the code and upload to your device and it should connect to your router.

I have had 3 devices connect to my network in this way with no issues. You just need to look in your router's device table to see what IP address is assigned to it.

jimhiggs, 2019-01-07:
Ahhh thanks @quantum8  for clearing the fog. Of course they are different wifi networks so I can leave them both at 192.168.4.1. and also thanks for the info regarding wifi setup. I have one working now but the other is still not connecting although I have not tried setting it up using the W10 laptop. I will give your method a go instead.

lalo-uy, 2019-03-11:
It could be possible to leave one in AP mode an the other as station connect to it?

grga12345, 2020-06-14:
which software to use to get to codes and set ssid and password, @quantum8 ?
I also have troubles with savin wifi data...