vitotai/BrewPiLess #158: error compiling from source

roguenorman, 2019-01-20:
Hi,

Im having ans issue building brewpiless from the master branch in platformio. When building it without any changes it comes up with this error:

src/WiFiSetup.cpp:72:41: error: 'DEFAULT_HOSTNAME' was not declared in this scope

I went into config.h and uncommented the following line: 

#define ESP8266_WiFi 1

That error has now stopped but I'm getting a new error: 

src/BrewPiLess.cpp:1408:17: error: 'connectClients' was not declared in this scope

I had a look and I cannot find where connectClient() has been defined. Am missing something?

Also the document here: https://github.com/vitotai/BrewPiLess/wiki/Software-Configuration states that there is a file called espconfig.h but I'm pretty sure that has been integrated into config.h. 

Thanks

roguenorman, 2019-01-20:
I just tried it on windows 10 and it builds without any errors.

petobena, 2019-01-23:
hi @roguenorman,

in WiFiSetup.cpp change the 4th line to #include "Config.h"

see my Pullrequest https://github.com/vitotai/BrewPiLess/pull/154/commits/51ee931e86a977f70d7e666d9f66b25e7caf3b70

roguenorman, 2019-02-04:
yeah didn't make a difference.  I fixed the 'DEFAULT_HOSTNAME' error by compiling it on windows (didn't work on Linux). then I commented out #define ESP8266_WiFi 1 which I assumed enabled WiFi but even without it, WiFi works.