vitotai/BrewPiLess #159: brewpiless not displaying ssd1306

Mailliwivad, 2019-01-24:
Newbie here.

I used the esp flasher tool to install BrewPiLess.v3.4r1.oled.bin. 
BPL works fine but I cannot get any display on the ssd1306, do I need to install a driver or something along with binary to get it to work?

Thanks for any help.
Dave

vitotai, 2019-03-01:
Supposedly, the OLED should just work. 
All my OLED lcds were burned out, so I need to order a new one.

quimnut, 2019-04-15:
Will the classic 3.5 image work with OLED? There is no specific binary for OLED in the 3.5 binaries like the older versions,

OLED on the 3.4 binary works, but I have the date bug.

Thank you Vito!

quantum8, 2019-04-16:
Yep, you just need to compile it yourself @quimnut

quimnut, 2019-04-16:
Oh yea, it was missing from the bins folder so I was curious if it was default or something now.

vitotai, 2019-04-18:
I don't have working OLED on hands. Therefore, I don't know if it works or not.

Mailliwivad, 2019-04-18:
Update: eventually I was able to get the old to work on 3.4, but it wasn't reliable for me. Sometimes it would work on startup but then cut out when I accessed the frontend over WiFi, sometime it wouldn't come on at all. Eventually I decided just to switch to the LCD version when Vito released 3.5. the display works well now.

Thank you for this project.