vitotai/BrewPiLess #160: Change hysteresis?

moscabeer, 2019-01-24:
Is it possible to increase hysteresis?
Example: put a differential of 0.3 ?? set temperature to 20, when rising to 20.3 will turn on the refrigerator, if it falls to 17.7 will turn on the heater.

Mine always calls 0.1. My refrigerator is turning on and off very often, I'm afraid it might burn.

terragady, 2019-02-08:
You can change that with json commands and pids

petobena, 2019-02-08:
@terragady how? can you help us? with some tutorial? :) Thank you

terragady, 2019-02-08:
All explained in the wiki.

ElChangoGuero, 2019-02-14:
Changing idleRangeL/idleRangeH for the difference to start heating/cooling. coolTargetL/heatTargetH to stop cooling/heating.
eg. j{"idleRangeL":0.3}

allthepies, 2019-02-14:
BPL has a compressor protection capability where it won't switch on/off until a certain time period has elapsed.  How often is your fridge turning on/off ?