vitotai/BrewPiLess #163: Time and date in graph v3.4 is not correct

Hammersoe, 2019-02-07:
After testing several units and different versions, I figured either I am doing something wrong, or the time and date for the graph is not pulled correctly on v3.4. Tested with v3.1 and v3.3r1 and they both show time correctly with same chip and setup.
When starting up the BPL it connects to my network and pulls the time and date which it shows me in the OLED display and if I choose Now (old UI) under Beer Profile it sets the time and date to current which is correct.
However the graph when I hover over with the mouse it shows 1970 in the different temp fields. This seems to be affecting the usage of time as a variable for the Beer Profile.
New UI is missing the Now button under Beer Profile, but this might have been intended?

Any ideas?

petobena, 2019-02-08:
@Hammersoe try v3.5
Changelog:

v3.5

    MQTT Remote Control
    Pressure transducer support
    Add iSpindel WiFi signal if available, "Now" button in profile editing
    "Fixed" incorrect time issue after restart.
    Add DNS setting for static IP

Hammersoe, 2019-02-08:
@petobena
Thanks for your quick reply! I had completely forgotten about the other branch trees, so I had no clue v3.5pre was available.
I downloaded the source, compiled it with newui-oled and it sure looks like it is fixed in v3.5:)

closed 2019-02-08 by Hammersoe