vitotai/BrewPiLess #165: Massive iSpindel temperature error

walldad, 2019-02-16:
I just finished building an iSpindel and linking it up my brewpiless setup. The gravity reading seems to coming through fine but the temperature is off by almost 100F. The temperature reads correct in the iSpindel AP portal. I can't find a correction factor in brewpiless anywhere.

vitotai, 2019-03-01:
That sounds like a Celsius vs Fahrenheit issue.

kynsi, 2021-01-10:
I'm having the same problem. My iSpindel temperature reads ok and shows up as correct Aux Temp, but although in the device setting it's set as Beer temp and displays the correct temperature in the graph window it was going from +1 to -1 degrees C and is now suddenly at 67 C.

vitotai, 2021-01-12:
The values in "aux temp" series are raw values, while the values of "beer" temperature are "filtered". 
I am not familiar with the filtering algorithm, but in my test, the beer temperature will converge to real value after a short time.

kynsi, 2021-01-12:
Yes I saw this happen earlier but on my most recent brew it didn't so i had to use fridge temperature.
Could it be esp32 related? I switched from esp8266 to esp32..

vitotai, 2021-01-12:
Connect to BPL, and open Javascript console. You can see the raw data from iSpindel when received.

kynsi, 2021-01-13:
The raw values are ok, it's the filtered value which stays around either +/-1 degrees or jumps to 67..

vitotai, 2021-01-13:
You might try modifying the filter parameters by JSON command.
beerFastFilt, beerSlowFilt, beerSlopeFilt
set all of them to zero.

The jumps might be result of disconnecting, when some reports missed for some reason and no reports received within 5  minutes.

kynsi, 2021-01-14:
I have a connect interval of 30s. Also if there's no connection it would mean the aux temp and SG wouldn't update but they are changed almost real-time.

vitotai, 2021-01-15:
What is your temperature unit on BPL? Fahrenheit or Celsius?
What is the temperature unit on iSpindel? F, C or K?

The related code of "wireless sensor" is so simple that it has less than 70 lines. Check TempSensorWireless.h if you read source code. 

Maybe you can copy the output of Javascript console, and let me try if I can reproduce it.