vitotai/BrewPiLess #166: Resizing file system

Hammersoe, 2019-02-24:
Not an issue as fault in software, but more a question to how.

Since the new frontend makes the OTA useless with the 2m2m filesystem on d1mini, I was looking for a way to utilize the full 16m storage capacity on the Wemos D1 Mini Pro as for example 2m14m filesystem
Skimming through the source code I cannot seem to locate what part to edit before compiling and if I should program with 16MB flash size with ESP8266flasher.

Any help in this regards is greatly appreciated.

vitotai, 2019-03-01:
Check these:

https://github.com/vitotai/BrewPiLess/wiki/Flash-Layout-and-New-Frontend

https://github.com/vitotai/BrewPiLess/wiki/2M-sketch-Space

Hammersoe, 2019-03-09:
Thanks for your feedback @vitotai !
For whatever reason I had issues using the OTA with different builds of the same version- but after changing the ld-file for 16m to fit my needs I could use OTA from a 4m2m build already running on a unit.
OTA works like a charm and I have plenty of filespace for logs internally.

closed 2019-03-09 by Hammersoe