vitotai/BrewPiLess #168: JSON commands from testcmd.htm not working in BPL 3.5

peteng72, 2019-03-01:
It seems the JSON cmds sent from testcmd.htm are not received by BPL. I copied the example from the wiki:
j{”tempFormat”:”F”}
and the the ”C” cmd to see the result, but the tempFormat parameter was still set to ”C”.

Hammersoe, 2019-03-09:
The command in the wiki might be incorrect. 
I testet this using j{tempFormat:F} instead of j{”tempFormat”:”F”} and it changes the value on my unit.
So in all simplicity, just remove the quotation marks and you should be good.

terragady, 2019-03-11:
also try to use lower "c" to see the results, the capital C might default your settings

peteng72, 2019-03-11:
> The command in the wiki might be incorrect.
> I testet this using j{tempFormat:F} instead of j{”tempFormat”:”F”} and it changes the value on my unit.
> So in all simplicity, just remove the quotation marks and you should be good.

You are correct. Removing the quotation marks solved the problem. The wiki should be updated.