vitotai/BrewPiLess #173: WifiSetupClass error

louis-germain6, 2019-05-07:
Hey, when building via platformio I'm getting an error in the WiFiSetup class.

It's saying DEFAULT_HOSTNAME was not declared in this scope, I tried to define a DEFAULT_HOSTNAME = "brewpiless" that allows me to build correctly the code.

My problem second problem is that I can connect to the Wifi but it never starts me up the webpage to configure my Wifi.

Thanks for your help, I can explain it better if there is something you don't understand

louis-germain6, 2019-05-08:
Thanks, it's working, it didn't load the homepage as I opened the navigator but whit the IP adress I could connect to it and set my wifi setup

Thanks a lot

louis-germain6, 2019-05-08:
That was it, it didn't load the setup page as soon as I open the navigator but with the IP adress I could enter and configure my Wifi!!
Thanks a lot

closed 2019-05-08 by louis-germain6

petobena, 2019-06-04:
@vitotai hi, i write the error long ago, https://github.com/vitotai/BrewPiLess/pull/154 , you should sometimes also the pull requests

vitotai, 2019-06-05:
@petobena 
Sorry about that.
I am not familiar with Github and didn't notice that.