vitotai/BrewPiLess #175: Beer profile step limitation

prylnissen, 2019-05-20:
I've tried searching for any documentation to support the limitation i'm expeeriencing in the number of steps I can use in a beer profile. 4 steps is the lilmitation I see today. Is this correct? Would it be possible to add a few more steps to support a more granular lager fermentation and conditioning?

vitotai, 2019-05-20:
How may steps will be enough?
It's in fact 7 steps including 3 ramping steps.

You can modify the the maximum and rebuild to get more steps.
https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/src/BPLSettings.h#L56

The frontend needs to be edited, too.
https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/htmljs/src/js/script-control.js#L2

However, the modification will offset the settings, result in incorrect settings.

prylnissen, 2019-05-20:
I guess 10 steps would suffice, as I need 4 steps +ramping steps.

I’ve little experience in coding and building sw, but I could certainly try 😊

Thanks for the swift response

Thomas

https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/src/BPLSettings.h#L56<https://eu-west-1.protection.sophos.com?d=github.com&u=aHR0cHM6Ly9naXRodWIuY29tL3ZpdG90YWkvQnJld1BpTGVzcy9ibG9iLzIwYTU5OTJmMjRmMzVlYjU3ZDQ4MzFkYjhmNjlkMDhlNTIzMmEwZjEvc3JjL0JQTFNldHRpbmdzLmgjTDU2&e=dGhvbWFzQHBpY2hsZXIubm8=&t=azJhVDBvYkpvdEpYcHM2d1dpU2dSZDZ6Z2pCRHhQQ2ZSNnlLeWNjS3lYND0=>

https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/htmljs/src/js/script-control.js#L2<https://eu-west-1.protection.sophos.com?d=github.com&u=aHR0cHM6Ly9naXRodWIuY29tL3ZpdG90YWkvQnJld1BpTGVzcy9ibG9iLzIwYTU5OTJmMjRmMzVlYjU3ZDQ4MzFkYjhmNjlkMDhlNTIzMmEwZjEvaHRtbGpzL3NyYy9qcy9zY3JpcHQtY29udHJvbC5qcyNMMg==&e=dGhvbWFzQHBpY2hsZXIubm8=&t=ZllEdUVrU0xuaGRFVWdhQWdpaTZEcFhwR3ZiVEFOMWRZRFNPUnVZQWtYUT0=>

nerdworks, 2020-03-22:
Hi, I have the same issue. Hopefully this will be fixed in the next release :-)

--
Jøran

pjpeartree, 2021-01-20:
Hi, @vitotai ,

Thank you for all your hard work on this project.

Having only 3 ramping steps is a limitation for me too.

Please, can you give more detail about the impact of changing the MaximumSteps?

PS: I have the skills to change the code and rebuild, but I would like to understand the impact of that change on settings.

Is it possible to remove this limitation, in the future, and just have a dynamic number of steps without a fixed number of steps?

vitotai, 2021-01-22:
The answer is in the source.

pjpeartree, 2021-01-22:
I see that the Settings struct will have another size,

https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/src/BPLSettings.h#L222

and the load function will be unable to load previous settings properly:

https://github.com/vitotai/BrewPiLess/blob/20a5992f24f35eb57d4831db8f69d08e5232a0f1/src/BPLSettings.cpp#L24-L29

Can you confirm that it only affects loading preexisting settings?

pjpeartree, 2021-01-24:
Hi,

I have done the changes above and still have the step limitation on the web interface.

I have rebuilt the HTML files, the gzip files, using grunt i18n, with the new size and still no luck.

I notice that the generated header files at src/data are still the same.
I know that the real page is compressed and embedded in the source code, in this header files. How can I re-generate these header files?

Thank you

vitotai, 2021-01-24:
I don't have time to write coding tutorial.
You might just need to "reload" the apge.
I've done the modification in this version,
https://github.com/vitotai/BrewPiLess/tree/4.2pre

pjpeartree, 2021-01-24:
Thank you @vitotai.
I have managed to generate hex and add manually to src\data\english_control_htm.h base on my change of MAX_STEPS.

I do not know why Platform.IO isn't rebuilding these embedded HTML header files automatically.

For my side, you can close the issue/enhancement.