vitotai/BrewPiLess #176: Temp control difference on two identical setups

prylnissen, 2019-05-20:
I am running two BPL setups for two separate fermentors. I'm using a glycol cooler and pumps to cool the fermentors. The BPL's have the same fw, and the settings have been set according to the recommendations for glycol setup. When I set both BPLs to constant temp 10 degrees celcius, the temperature curves differ a bit. One tank seem to constantly let the temperature rise to approx. 11 degrees before cooling. The other tank seems to start cooling at 10.1 degrees. I've studied the settings, and I can't seem to find any differences that would cause this difference in behaviour.

[image]
[image]

vitotai, 2019-05-21:
What is the vales of idleRangeH for both?

prylnissen, 2019-05-21:
The settings are "idleRangeH": 1.000,"idleRangeL":-1.000

Am I correct to think that these values means that the temperature will vary between +1 and -1 degree from the set temperature? If so, this is an easy fix, I guess

Will the values coolTargetH/L and heatTargetH/L have an impact on this behaviour as well?

vitotai, 2019-05-21:
Cooling will start when    [Current Temperature]  > [Target Temperature]  + idleRangeH.
The timing of cooling stop is a little more complicated. In a nutshell, the algorithm will accumulate an error for estimation of when to stop.