vitotai/BrewPiLess #177: Error in master compile - BPLSettings.cpp

DenisAntoine, 2019-05-21:
Hi,
I've got the following errors when compiling the master in plateformIo.
repeated multiple times :

BPLSettings.cpp
invalid use of incomplete type 'struct tm'

Thanks for your help

vitotai, 2019-05-22:
Add the following line at begging of BPLSettings.cpp
`#include <time.h>
`

DenisAntoine, 2019-05-22:
Hi,
I've added the #include <time.h> line in the BPLSettings.cpp but still it doesn't compile.
Time  1.5 library is installed in my plateformIo

Here's an extract of the errors log.

Best

src\BPLSettings.cpp: In function 'time_t tm_to_timet(tm*)':
src\BPLSettings.cpp:299:18: error: invalid use of incomplete type 'struct tm'
seconds= tm_time->tm_year*(SECS_PER_DAY * 365);

^
In file included from c:\users\antoi\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\cwchar:44:0,
from c:\users\antoi\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\postypes.h:40,
from c:\users\antoi\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\iosfwd:40,
from c:\users\antoi\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\memory:72,
from C:\Users\antoi\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:24,
from src\BPLSettings.cpp:4:
C:\Users\antoi\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/wchar.h:56:8: error: forward declaration of 'struct t
m'

vitotai, 2019-05-23:
What is "Time 1.5 library"?

<time.h> is basically Ansi C/C++ standard, or something like.
It is included in the tool-chain. You should find it in the same directory as wchar.h.

Try open the directory
C:\Users\antoi.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include/

And check "time.h" is there.

I don't have such issue totally have idea how to solve this problem. (Mayeb unstall the mysterious Time 1.5 library?)

DenisAntoine, 2019-05-23:
Thanks
I've uninstalled the conflicting library in my local envt and it went through the build without errors.
Now I've to discover more about the capabilities of the program.
Thanks a lot for your help

closed 2019-05-23 by DenisAntoine