vitotai/BrewPiLess #179: Fridge Temp reports identical values to Beer Temp

sryburn, 2019-06-01:


vitotai, 2019-06-01:
Check your fridge sensor.

To support Glycol, "fall back sensor" is enabled. The fallback thing is that fridge temp will be from beer sensor when fridge sensor is missing.

Check the wiki for detail:
https://github.com/vitotai/BrewPiLess/wiki/Use-with-Glycol

sryburn, 2019-06-01:
Thanks @vitotai, the fallback makes sense now! I will try another sensor or maybe disable the settings for glycol support and see what happens.

sryburn, 2019-06-03:
I have concluded that my temperature sensors work fine by themselves. My issue was resolved by disabling the glycol support settings in `Config.h`, so the simple solution is to only enable glycol support if you actually need it. It still could be something to do with my sensors I guess, eg maybe they don't always return a reading.

vitotai, 2019-06-03:
That happens.

I had to rewire my sensors,  ditching DIN connectors, to solve this issue.

closed 2019-06-04 by sryburn