vitotai/BrewPiLess #182: Support for Tilt hydrometer

exige81, 2019-06-11:
Im not sure offhand if there are specific hardware limitations, but I'd be very interested if BrewPiLess were to support tilt hydrometers.   I see that another project [tiltbridge](https://github.com/thorrak/tiltbridge) uses an ESP32 to pull the data.

vitotai, 2019-06-12:
TILT uses BLE to broadcast information.
ESP32 supports BLE while ESP8266 doesn't.

splitice, 2019-09-10:
This could be done via tilt pi.

https://tilthydrometer.com/products/tilt-pi-v2-buster-raspberry-pi-sd-card-image-download

vitotai, 2019-09-10:
Using anther device as bridge is a simple solution.
For example, if BPL accepts Fermtrack format, TiltBridge can be used.

splitice, 2019-09-10:
Heres a very basic implementation of the TiltPI format.

https://github.com/splitice/BrewPiLess/commit/47f434e55fa55e2d8a6fab69f1c54077ae8b1acf

Data format is as follows: 
```
Timepoint=43718.78153083333&Temp=64.0&SG=1.019&Beer=Untitled,14375&Color=RED&Comment=
```

TODO:
 - test
 - doesnt support `Timepoint=`
 - support multiple tilt devices (`Color=`)

splitice, 2019-09-10:
One observed bug - the wireless temperature sensor is not receiving the temperature reading from setAuxTemperatureCelsius. I am not sure why, doesnt appear to be anything to do with the very basic code I added. @vitotai ?

vitotai, 2019-09-10:
If the request is GET request, in which the data/parameters are included in the URL, the statements are wrong.
`if(!request->hasParam("SG", true) || !request->hasParam("Temp", true)){
`

The hasParam("SG",true) checks "POST" request.

exige81, 2019-09-10:
Fetching tilt data from another device like tiltpi is a good intermediate solution.

Ultimately, powering it all from one device would be a better long term solution.  To that end i'm currently working on getting a version of BrewPiLess working on ESP32 (#128).  Making some decent progress so far, currently trying to figure out why the wifi scan only populates the modal sporadically on the config page, but will have more time to dig into it next week. 

Either way, any progress in getting the tilt integrated is good.  :thumbsup:

splitice, 2019-09-10:
@vitotai

It's post (urlencoded). The committed code works.

The only issue is with the Wireless Temp Sensor check which doesn't seem to work (does it currently work with iSpindel's?)

@exige81 yes that would be better.

vitotai, 2019-09-11:
If you see AUX temperature line on the chart, it's working.
It's now easy to check whether Wireless Temp Sensor works or not.
The reading on Device Setup is retrieved only when "listing". The readings from Wireless sensor only last for 5 minutes.

splitice, 2019-09-11:
Ah nice, looks like it is working. My temperature is just too stable :)

A really nice feature for iSpindel's and Tilt's would be to use them as Beer Temp with a Wire sensor as a backup. i.e Tilt should not be outside 3deg of wired sensor.

The reason for this is that if the external sensor used as beer sensor is disconnected while temp is low, the control loop would continue to heat (or cool if high) - resulting in a overheated (or cooled) ferment. Correct?

Thoughts?

vitotai, 2019-09-11:
The reason that the temperature from external sensor is valid for only 5 minutes is to avoid the situation you mention.
If the external sensor doesn't report temperature within 5 minutes, it is regarded "disconnected" and the temperature control will STOP.

splitice, 2019-09-11:
Hmm nice idea. I should check how many degrees of change I get over 5 minutes, I suspect it would be too much. I'm heating using a 10A (2400W) hair dryer in the fermentation chamber.

What would your thoughts on a temperature guard be if a second beer sensor existed?
Also what happens if two sensors of the same function are defined normally?

vitotai, 2019-09-11:
The temperature control part is directly from BrewPi, and I didn't change much of it.
To be exact, I only change code to allow configurable minimum cooling/idle time.

I have thought of something you mention,  but what is the point to use external, maybe unreliable, sensor when you have a connected sensor?

splitice, 2019-09-11:
The connected sensor outside the fv. Something like a tilt or iSpindel is inside the fv.

vitotai, 2019-09-26:
FYI, ESP32 porting is almost done.
However, BLE is a big problem. First of all, the library is so HUGE. 750k+ program size increases by Including the library. More over, Bluetooth library takes too much heap to run other tasks.
A solution, NimBLE is coming. However, it's not available right now. We will have to wait for it.

andycwk, 2020-07-08:
Hay @splitice what's the status of this ticket.. are you likely to be raising a PR for this excellent start (assuming PRs are welcome from the community)?

kall3, 2020-07-14:
Thinking of changing from ESP8266 to ESP32 for my build... What is the status for integrating NimBLE and adding support for tilt?

vitotai, 2020-07-17:
NimBLE runs and scans. It's progressing slowly. I don't have a TILT, so I can only test it by using simulating APP of iPhine. I don't know if it works or not for real TILTs.

kall3, 2020-07-17:
Once I get my hands on a ESP32 board I'll give it a go to see what kind of data we get out of my TILT using NimBLE.

vitotai, 2020-07-17:
I don't have a TILT, as mentioned. May I ask you what settings are needed for TILT. Gravity offset? Temperature offset?

andycwk, 2020-07-17:
Hay @vitotai Looking at node red for the TiltPi the tilt BT beacon sends out the temp & SG readings pre-calculated via the tilt along with the device colour uuid
[image]
.. so you just need to consume the info

sedgington, 2020-07-17:
I have a Tilt and an ESP32 Running BPL master branch. What steps do I have to do to enable and test Nimble?

vitotai, 2020-07-19:
The data format is clearly documented in this page. https://kvurd.com/blog/tilt-hydrometer-ibeacon-data-format/

A trial version will be pushed in a few days.

vitotai, 2020-07-20:
> I have a Tilt and an ESP32 Running BPL master branch. What steps do I have to do to enable and test Nimble?
> […](#)

Please try this branch, and let me know the result.
https://github.com/vitotai/BrewPiLess/tree/Nimble.trial

You have to build by yourself. BTW, the partition changes, and you can't flash OTA.

sedgington, 2020-07-20:
I tired building it and it failed due to the partition change. Do I need to load something else into PlatformIO? It appears from the documentation that the .ini is set up correctly but I get the following errors when trying to build the nimble branch:
*** [.pio/build/esp32/partitions.bin] Source `partition.csv' not found, needed by target `.pio/build/esp32/partitions.bin'.
src/WiFiSetup.cpp: In member function 'void WiFiSetupClass::begin(WiFiMode, const char*, const char*, const char*, const char*)':
src/WiFiSetup.cpp:126:15: warning: variable 'status' set but not used [-Wunused-but-set-variable]
   wl_status_t status;

sedgington, 2020-07-20:
so it looks like I need to define partition.csv and put it in the same hidden folder with default.csv. Which on my Mac is:

~/.platformio/packages/framework-arduinoespressif32/tools/partitions

If you can give me the definitions, or tell me where to get them, I can copy the defalut.csv, rename the copy partitions.csv and then enter the appropriate values. Thanks.

vitotai, 2020-07-21:
sorry. I forget to "git add" the partition.csv. Please get it from the branch. Just put it in the same directory as platformio.ini.

sedgington, 2020-07-21:
Yeah that was the problem Built and uploaded with no problems. I went to “Gravity” and enabled my Tilt as black and submitted. What I missed several times on that page was the slide button on the far right you need to push to turn the tilt reading on., (perhaps put the slide button to the left of the tilt color selection?) In any case, once I did that, I went back to the “Graph” tab and instantly had specific gravity and temperature from the tilt—so it does work! I checked it with my iPhone Tilt2 app and it was giving the same readings. I guess the next step would be calibration as with the iPhone app...

Finally, I’m wondering what the red dot next to “Final Gravity State” represents? FYI, my tilt is in water.

Great job. Thanks!

vitotai, 2020-07-22:
Great! What kind of calibration is needed? Input a few pairs of gravity, or like BrewNCal, calibration while brewing?

sedgington, 2020-07-22:
I'm thinking first of all that you can offset temp or specific gravity based on a instant read thermometer or SG of water--this capability is in the Tilt2 app on iphone. From that point you could 'tune' the SG reading as BPL does with iSpindel or as outlined by tilt using sugar and water mixtures.

vitotai, 2020-07-29:
I have to charge my iSpindel between brews. Every time I open the cap to charge it, the center of mass moves, and re-calibration is needed. Thus, I design brew-N-Cal to calibrate iSpindel during fermenting. Tilting angles are stored in the log files, as well as the calibration data. Log viewer derives the formula on the fly and displays gravity.

The story seems to be different for Tilt hydrometer, since the battery seems to last way longer. Some people even claim that calibration is not necessary.

Based on the information I get so far, calibration of TILT should survive through brews, and the calibration data should not be stored in the log files.

andycwk, 2020-07-29:
Hi @vitotai I would agree with that ... probably good enough to store calibration capability at a global level for each tilt colour :)

I'm just in the process of building out a new kit with the ESP32 to replace the inkbird so while I might not be able to tweak the code I would be more than happy to do some testing for you once I have the bits installed in the fridge :)

andycwk, 2020-11-19:
Hi @vitotai just checking in...

I've finally found time & money to finish my build and decided to give the Tilt logic a spin!

I noticed that there are two trial branches dedicated to tilt... the one mentioned a while back on this thread `Nimble.trial` and then `Tilt.trial`. Can I just check which one I should be running with.. from the commit history Tilt.trial appears the fresher one - and the one I've deployed - however it doesn't seem to be seeing my blue tilt :/

Any thoughts?

vitotai, 2020-11-19:
Tilt.trial is the latest branch.

I don't have a Tilt, and I use my iPhone to simulate a Tilt by editing the iBeacon. I have no idea what happed to you.

andycwk, 2020-11-19:
Hi @vitotai 
No problem, I noticed some commented out debug stuff in the scan logic... I'll uncomment those and diagnose - I'll also feed back as needed/create PR with any changes - cheers 👍

andycwk, 2020-11-19:
Hi @vitotai after a few cold hours in the garage (tilt already in a brew) and a bunch of diagnostics in hand.. I've identified the issue... a minor typo in ExternalData.cpp - apart from that, it works like a charm 💯 

I've raised a PR if you want it 👍 

https://github.com/vitotai/BrewPiLess/pull/281

edsenabr, 2021-05-16:
I'm brand new to BrewPiLess, have just compiled and installed it on my Lolin D32, and my plan is to rely only on a Tilt for fermentation control, w/o any wire getting into the fridge. But it looks like Tilt is only being used as an auxiliary thermometer, and its data is not being actively used for triggering the actuators. Did I get it right ?

vitotai, 2021-05-17:
Read some documents, if you would. You have to set "external sensor" as beer temp.

felipempbr, 2022-02-23:
Hi @vitotai,
I'm trying to use the Tilt Hydrometer with the brewpiless on ESP32, but the Density and Temperature readings are out of calibration.
The Tilt Calibration page doesn't work for me. The "ADIC." button does not result in any additions to the table.
What could be happening?
[image]