vitotai/BrewPiLess #183: Remote logging stops

hulzer, 2019-06-21:
I run latest version 3.5.1 on Wemos D1 Mini.
I have connected iSpindel to "brewpiless" AP. Brewpiless is also connected to my local Wifi router.
I setup remote logging to a node-red instance via http POST and then writing to Influxdb. 
Sometimes the logging stops after running many hours (>10). I have no data points anymore in the logging database.
When I connect to brewpiless UI, is see that the latest iSpindel update is long ago. But now, the logging starts to work again. I just accessed the UI.

It seems that the logging stops after 12 hours...

[image]

Do you have any idea? Thanks.

vitotai, 2019-06-22:
I run two BPLs. One of them disconnects from LAN after 3 ~ 5 days, while the other keeps working good. I don't have any idea. Also, sometimes, iSpindel skips reporting once or twice. Debugging network issue is really hard, especially when we don't know when it will happen.

hulzer, 2019-06-22:
I have two BPLs, one nodemcu and one d1 mini.
 Is there any difference in flashing?

At the moment, I reflashed both BPLs but now none connects to Wifi. I use Station+AP mode. Sometimes I can connect to AP but it's very unresponsive, sometimes I get a timeout.

I tried to get some output on serial console but the console doesn't show anything!? I this normal? I connect to COM port but I don't get any message displayed. Do I need to enable something? 

Thanks.

quimnut, 2019-06-23:
I run local and remote logging on 2, (v2.6 and v3.0) POST to node-red and then send a push to my phone if there is a > 0.2c difference in set and current beer temp, the remote logging has never failed me, the node-red graphs are always current.

This went off last week on the v2.6 unit i've run for a couple of years, it started clicking the relay and resetting a _lot_. I moved the relay to 5v which made it stable for a day but I've also had to disable local logging to keep it running for a few more now, I suspect the both issues could be due to the esp's 3v reg failing.

Do you have a local logging on, are you seeing the R flag at the bottom much?

I would use IP addresses not dns names in the logging config.

For general instability I work through;
Good usb adapter, not 500mW and a good cable, some cheap or old low awg cables are just no good. 
Power the relay off the 5v pin anyway, the 3v reg on esp's is often only rated at 850mW
Erase the esp, re-flash and run format-spiffs.
Failing this try a different esp, i've had to throw a few out over the years due to general flakiness, they are cheap, I buy cheap ones too.

vitotai, 2019-06-24:
Yes. SPIFFS must have something to do with stability.
ESP8266 is a tiny system, and the file system might not be good  enough to run local logging. My setups that don’t have local logging enabled seem to be more stable.

hulzer, 2019-06-24:
I got it working again! Thank you.
I erased whole flash and formatted SPIFFS.
The device responds much more smoothly than ever before. 

Thanks!

closed 2019-06-24 by hulzer