vitotai/BrewPiLess #184: Navigation to "Other URL"

Fcsteve777, 2019-07-02:
Im sure this is me and not the software. I have 2 units up and running 1 named left and 1 named right.

I cant seem to figure out how to navigate to the "Other Url" pages for configuration in 3.5

When i am accessing thru my home network i have tried HTTP://"Network IP Address"/LCD
Connected to the esp directly i have tried HTTP://"DSP's IP Address"/LCD as well as brewpiless.local/LCD

Just a blank screen is all i get, Any pointers?

allthepies, 2019-07-02:
I think it's been removed in the latest versions of BPL.  I used to access the "LCD" page and it now doesn't work, as you're also finding.

vitotai, 2019-07-03:
I am sorry. I thought nobody was using. It turns out I need it, too.
It was back and revised in 3.6. I also make it auto resized, though not perfect.
You can try 3.6pre branch, but please note the flash layout is default to 4m2m. Flashing via USB is recommended if flash layout changes.