vitotai/BrewPiLess #187: ERROR: send_offer (error -13)

Naesstrom, 2019-09-03:
I flashed the latest version today on a nodeMCU and it boots up to AP mode and I can connect.
I don't get a ip adress tough and if I change my IP manually to 192.168.4.2 I can't connect to BPL.
Log shows following:
```
Setup()
SPIFFS.being() Success.
Init LCD...
LCD Initialized..
syscfg:0, timeinfo:156, gdc:168,                tempSchedule:204, brewStatus:300, logFileIndexes:320,             remoteLogginInfo:636, autoCapSettings:1080, parasiteTempControlSettings:1092
Starting WiFi...
begin:,SSID: pass: IP:(IP unset), gw:(IP unset)

Saved SSID:""

AP mode:3, used;3

create network:brewpiless pass:brewpiless
WiFi Done!
connect to Time server
Time from NTP :28801
failed to connect NTP, load time:300
HTTP Update server started

HTTP server started
SPIFFS space:1936898
start volatiel log
mode ␀ => o
startVolatileLog, mode=o, beerteemp=32767
**disc:,SSID: pass: IP:(IP unset), gw:(IP unset)
Lost Network. auto reconnect 1
add 1
aid 1
station: 38:f9:d3:62:ab:c1 join, AID = 1
ERROR: dhcps send ack (error -13)
ERROR: dhcps send ack (error -13)
ERROR: dhcps send ack (error -13)
ERROR: dhcps send ack (error -13)
ERROR: send_offer (error -13)
ERROR: send_offer (error -13)
etc.
```

vitotai, 2019-09-03:
What a coincident, I just found this issue and solving. Wait for v3.6.1.

Naesstrom, 2019-09-03:
hehe, great!
yeah no stress for me, still waiting for the relay shield that I ordered today! 😺

vitotai, 2019-09-03:
try v3.6v2. just pushed.

Naesstrom, 2019-09-03:
```
Error: Invalid '/Users/erik.nasstrom/Downloads/BrewPiLess-master 2/platformio.ini' (project configuration file): 'While reading from '/Users/erik.nasstrom/Downloads/BrewPiLess-master 2/platformio.ini' [line 50]: option 'lib_deps' in section 'env:development-esp8285' already exists'
```
Commented out line 50 and it compiles but didn't seem to boot correct after flashing.

Naesstrom, 2019-09-03:
Scrap that, just me that hadn't changed to development instead of sonoff! Working now. Making a PR with some minor changes that platformio was complaining about!

vitotai, 2019-09-07:
Development is used by ME during development. Don’t mess with that.

closed 2019-09-19 by Naesstrom