vitotai/BrewPiLess #189: iSpindel questions with Brewpiless

ebar03, 2019-09-10:
I have a question on how to use the iSpindel with Brewpiless and wondering how some one you set yours up.
Am I supposed to use the DS18 sensor with the iSpindel or just the iSpindel alone?
If I set the External Sensor (iSpindel) as Beer Temp, should I remove the other temperature device slots or keep one as Chamber Temp and change the other to Room Temp?
What reporting/interval period is best for the iSpindel? Wiki said nothing more than 5 minutes, but how work that affect battery life.
Appreciate the project and all the help.

vitotai, 2019-09-11:
If you are using a freezer/fridge as fermentation chamber, a DS18B20 sensor as Room Sensor is necessary.

I personally haven't tried using iSpindel as Beer sensor. I implemented it because Sam, the creator of iSpindel, requested. IMO, less than one minute report period might be necessary.

You can find some information here:
https://github.com/vitotai/BrewPiLess/issues/5

vonfritz, 2019-11-07:
I think that at present it is not possible to use ispindel as a temperature detector, interval time under 5 minutes consume a lot of battery power.