vitotai/BrewPiLess #190: Failed to Connect to server Controller not updating

Raoulfonz, 2019-09-22:
Hello 
I have spent some time trying to get this working and hope you can point me in the right direction as Im stumped
I was not able to get to 192.168.4.1 on V3.6 thorrak or V3.6v2 thorrak although I can connect to AP. 
I can however connect and config successfully with V3.5thorrak.
Have built Thorrak's board with temp sensor which reads correctly in BPI, and have been stopped by this error as I  cannot seem to find any assistance with that error. Its probably really simple
Any guidance is appreciated.
Thanks - Raoul

[image]

quimnut, 2019-09-22:
Flash 3.0 or less before the wifi manager changes were added, get it on the network, then upgrade.

quimnut, 2019-09-22:
I have tz issues on the newer versions using the built in logs, and as the esp can't cope with 2 connections, it dies when someone comes over that has been working on theirs and their phone connects automatically, then complain my wifi is down :)  when they are infact connected to the BPL. The solution to this is run 3.4 where you can have only station mode.

vitotai, 2019-09-22:
If BPL has connected to a WiFi network before upgrading without change of flash layout. It is exact this behavior that stop me from finding the issue.
The "Auto Reconnect" feature in framework 2.2.0 seems to result in disconnection of AP mode if BPL cannot connect to an AP in AP_STA mode.
The reconnection must be redesign, and the 36v2(well, should be r2, I was in a hurry.) is just a quick solution.

quimnut's advice is valid only when the same flash layout is used, which should be 4m2m.

vitotai, 2019-09-22:
Update:
I just push a 3.6r3. You might try it.

Raoulfonz, 2019-09-23:
Thank you for your help
I flashed 3.6r3 version. Now I can get to 192.168.4.1 and behavior is same as version 3.5... so that is now working.... good.
Problem is the behavior is the exactly the same as 3.5, and I am still getting the "failed to connect to server" error as before
[image]
Here are my flash parameters FYI
[image]
[image]
I also flashed a new wemos and behavior same.
Thanks for any guidance

vitotai, 2019-09-23:
If you can use Chrome and open Javascript Console(Alt+Ctrol+J on Mac), under menu "View/Developer options", copy the log for me.

Raoulfonz, 2019-09-24:
Hope this is what you require:

Violation] Added non-passive event listener to a scroll-blocking 'touchstart' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Dygraph.addEvent @ dygraph-combined.js:3
Dygraph.addAndTrackEvent @ dygraph-combined.js:3
t.createDragInterface_ @ dygraph-combined.js:3
t.__init__ @ dygraph-combined.js:2
t @ dygraph-combined.js:2
BrewChart.createChart @ index.htm:1
BrewChart.updateChart @ index.htm:1
i.onload @ index.htm:124
dygraph-combined.js:3 [Violation] Added non-passive event listener to a scroll-blocking 'touchmove' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Dygraph.addEvent @ dygraph-combined.js:3
Dygraph.addAndTrackEvent @ dygraph-combined.js:3
t.createDragInterface_ @ dygraph-combined.js:3
t.__init__ @ dygraph-combined.js:2
t @ dygraph-combined.js:2
BrewChart.createChart @ index.htm:1
BrewChart.updateChart @ index.htm:1
i.onload @ index.htm:124
index.htm:124 [Violation] 'load' handler took 152ms

Thanks

vitotai, 2019-09-24:
Use Chrome from Google. It's available on every platform.
I don't use Firefox and don't have it.

lbussy, 2019-09-24:
@vitotai if I am understanding this correctly, you need a way to be able to interrupt the auto-connect if there are invalid credentials?

I used a [Double-Reset Detector](https://github.com/jenscski/DoubleResetDetect) (as well as a pin, as well as an application selection ... three is better than one!) to clear the WiFi settings and present the configuration portal.  Actually, I think except for the application-based one I just pop the portal without clearing the config in case it was done in error.

Raoulfonz, 2019-09-25:
The text above the "Firefox shows this" is from chrome as requested. 
I Just thought I would add that too. sorry for confusion.

vitotai, 2019-09-25:
Those messages are strange to me.
What I am familiar with is simply the "console", which will show errors if any.
[image]

vitotai, 2019-09-25:
> @vitotai if I am understanding this correctly, you need a way to be able to interrupt the auto-connect if there are invalid credentials?
>
> I used a [Double-Reset [...]

That's a sound suggestion. I've thought of buttons because RESET pin isn't always accessible.

Raoulfonz, 2019-09-25:
Here is a screenshot. Thanks for you attention and assistance.
[image]

allthepies, 2019-09-25:
Sure this isn't a websockets vs SSE config issue ?   i.e. UI using websockets but server (BPL) configured for SSE ?   I had this initially with one version before I found the config compilation directive to enable websockets in BPL.

vitotai, 2019-09-25:
The log said "Connected", so it should not be SSE/WebSocket issue. (BTW, SSE might not work for now.)

The error messages show ALL three error at the same time, which implies the Javascript couldn't run to certain point. I would guess it's some kind of security setting issue. Th "Violation..." messages also implies that there are some kind of security settings that mine doesn't have.

Raoulfonz, 2019-09-25:
If it helps the error occurs on my phone or other PC as well.
What do you use to flash with and what are the flashing parameters. Worth investigating?

vitotai, 2019-09-26:
I use PlatformIO on Mac.
Would you try http://192.168.1.203/testcmd.htm
and capture the screenshot with Javascript Console?

The script somehow stops but the console shows warning that I've never seen.

Raoulfonz, 2019-09-26:
Here you go
[image]

vitotai, 2019-09-26:
The connection between the browser and BPL seems fine, but there's something strange.
The message

C{"tempFormat":"C"...}

is sent by demand. It should not be there you if you don't send a "c" character(command).

I don't get it and have no idea why.

Regarding the main page, the "warning" might be the reason or the same cause, but I don't know why they show on your browser but not mine.

Raoulfonz, 2019-09-26:
I reflashed using a different tool nodemcu pyflasher.
This is in console.
[image]

Raoulfonz, 2019-09-27:
Here is screenshot
[image]

vitotai, 2019-09-27:
The BPL controller and web socket connection are working fine. It's just the Javascript doesn't run for some reason.
Maybe you can try another browser.

lbussy, 2019-09-28:
> If BPL has connected to a WiFi network before upgrading without change of flash layout. It is exact this behavior that stop me from finding the issue.
> The "Auto Reconnect" feature in framework 2.2.0 [...]

Okay, I finally hit this, which is described [here](https://github.com/tzapu/WiFiManager/issues/866) I believe.  I don't see much action on it.  Can you explain how you got around it?  I was really hoping not to have to re-write my own libraries. :)

Raoulfonz, 2019-09-28:
It has same behavior Chrome, Firefox Opera and moble chrome.
and on different PC (windows7.
Interesting error from Firefox:
Source map error: request failed with status 404
Resource URL: http://192.168.1.203/dygraph-combined.js
Source Map URL: dygraph-combined.js.map
It advises to learn more at
https://developer.mozilla.org/en-US/docs/Tools/Debugger/Source_map_errors?utm_source=mozilla&utm_medium=firefox-console-errors&utm_campaign=default.

vitotai, 2019-09-28:
> Okay, I finally hit this, which is described [here](https://github.com/tzapu/WiFiManager/issues/866) I believe. I don't see much action on it. Can you explain how you got around it? I was really [...]

I check if any saved WiFi network exists by checking WiFi.SSID() at boot up. You can called "wifi_station_get_config()" to get stored SSID instead. If there is no saved network, AP mode is entered without starting STA mode. WiFi.setAutoConnect(false) is called after 5 seconds in connection attempt if it fails. Furthermore, WiFi.begin() seems necessary along with WiFi.setAutoConnect(true) to try reconnect().

I have to admit that I don't really understand how WiFi works from the beginning. For example, the system crashed or didn't work as expected SOMETIMES when changing mode with older version of framework, like two years ago. Adding WiFi.mode(OFF) before change mode seems to solve it or at least reduce the chance of crash. I also found that WiFi behaves differently in different version of framework, like "setAutoConnect()" issue.

vitotai, 2019-09-28:
> It has same behavior Chrome, Firefox Opera and moble chrome.
> and on different PC (windows7.
> Interesting error from Firefox:
> Source map error: request failed with status 404
> Resource URL: [...]

No. It's different. There is no such error in all the pictures you provided.

The error is caused by different versions between front-end(HTML/JS) and controller code. You should "REFRESH" to force reload the page.

Raoulfonz, 2019-09-28:
That is true as I sent only pictures of Chrome as you requested.
When I was checking different browsers, I went into Browser Console. instead of Web Console in Firefox, and that was a new message I saw. I thought it may be of help
Sorry Vito I wont mention Firefox again   :-)
The behavior I referred to was the following error
[image]
Have refreshed and I get same error on 2 other PCs W7 & W10, and my Android mobile
The one I am on now is W8.1 (hope that helps)

vitotai, 2019-09-28:
I can't help in you case.

lbussy, 2019-09-28:
> I check if any saved WiFi network exists by checking WiFi.SSID() at boot up. You can called "wifi_station_get_config()" to get stored SSID instead. If there is no saved network, AP mode is entered [...]

At this point, it seems like even if I "ERASE FLASH" I can't get the AP up.  Curiouser and curiouser .  I'll have to make a minimum config and step through it ... again.  :)

lbussy, 2019-09-30:
@vitotai just an FYI:  I have moved to the "Development" branch of WiFiManager and have not been able to reproduce the original issue I was having.  I know you've already merged/refactored but I thought I'd let you know.

Raoulfonz, 2020-01-03:
SOLVED
The ESP8266 must be plugged in to circuit board with sensors and then flashed.
I was flashing just ESP8266, then plugging into breakout board.
3.6r3 working well.

closed 2020-01-03 by Raoulfonz