vitotai/BrewPiLess #191: Build error with Wifi control enabled

wpeterw, 2019-09-24:
With:
#define ESP8266_WiFi 1

I get a build error:

src\BrewPiLess.cpp: In function 'void brewpiLoop()':
src\BrewPiLess.cpp:1546:17: error: 'connectClients' was not declared in this scope
  connectClients();

I want to control Brewpiless with Fermentrack over wifi. I guess this is the correct setting ?

vitotai, 2019-09-25:
You must be mistaking BrewPiless with thorrak's brewpi-esp8266:

https://github.com/thorrak/brewpi-esp8266

The one most important point of BrewPiLes is Pi Less, or running without a RPI. The porting to BrewPiLess is done by thorrak for brewpi-esp8266, and that's why you see this option. I will remove that some time later.

wpeterw, 2019-09-25:
No, this is from your repo. Line 434 in Config.h

vitotai, 2019-09-25:
Yes. I see. I should've removed that.

You should go to brewpi-esp8266 if you want to work with Fermentrack. That's what I meant.