vitotai/BrewPiLess #193: Temp control not stable after a few days.

arjepsen, 2019-10-26:
Hey.
I have a problem with the temperature control of my setup.
For the first couple of days, the temperature are kept very stable, but then the controller starts both heating and cooling a lot, make the beer temperature move around a lot more. Weird, since this is after the main fermentation seems to be finished.
Here's the current graph:
[image]
As you can see the temperature seems very stable, untill close to noon on 25th.
Can anyone tell me what's going on and how to make it more stable?

quimnut, 2019-10-26:
Move your chamber sensor under the cooling element of the fridge to reduce the overshoot.

arjepsen, 2019-10-27:
Not sure I understand - you mean I should move the chamber sensor, so that it's reacting quicker to the cooling surface of my freezer?

quimnut, 2019-10-28:
Yes, but looking more closely today you get a lot of overshoot of heat from the minimum heat time, not sure thats easily remedied, but lowering overshoot on cooling might help on one side.

You can check out what the overshoot estimator says here and adjust.
https://github.com/vitotai/BrewPiLess/wiki/JSON-command