vitotai/BrewPiLess #196: Virtual heating / cooling devices

brendongough, 2019-11-08:
The purpose of this feature would be to add http endpoints for local wifi or hard wired relay type devices. 

With the increasing amount of Sonoff-Tasmota devices, this cold be achieved by sending a http request to the device itself or an ifttt type request to a management interface.

This would hopefully allow increased adoption of the platform by removing the requirement for people to do mains voltage wiring. It does increase the reliance on a good wifi connection in order to function.

Thoughts on the potential feature?

lbussy, 2019-11-08:
I think I'd personally prefer a hard-wired relay over something that's most critical application design point ot turning on and off lights.  

I did just recently learn that in some countries, wiring line voltage (your "mains") must be done by a licensed electrician.  I suppose if that were the case in the US I might think about it differently, but then again I pretty much do what I want so ... it would just be another rule I break :)

ithinkido, 2019-11-09:
Hard wired will always be more reliable than WiFi. That said, Tasmota does send a reply to http commands which is would be a possibility to use as a control to check if the command has been received ( and hopefully completed)

vitotai, 2019-11-10:
It can be done without much difficulty but lousy.

What lbussy said is the most concerned. The design in question will introduce many weak links which include the deivce, the access pointer/router, the connection between BPL and AP/router, and the connection between the device and AP/router. If MQTT is involved, the server adds up.

NOT only reliability, compatibility would be an issue. I have been told that BPL has networking issue in Brazil or some SA country where the router/AP is controlled by the internet provider and users have no control over.

Therefore, although I have thought of it, I am reluctant to implement.

brendongough, 2019-12-13:
I'm sure it could be implemented with some *at your own risk* type labels or an advanced feature to enable at your own risk. It will help decouple a low voltage controller from 120/240v and use approved Tasmota switches. Personally, I have stable mesh WiFi at home and use these regularly to operate various things around the house in an automated fashion, without any issue.