vitotai/BrewPiLess #197: Saving config settings

RonnyRusten, 2019-11-10:
I'm trying to save Configurations, for example Title.
When I hit "Submit Query", the device restarts, and when I reconnect, Title is back to BrewPiLess
Is there anything I'm doing wrong?

vitotai, 2019-11-10:
What kind of power supply are you using?

RonnyRusten, 2019-11-11:
I use the USB cable only. I haven't connected it to anything yet, I was just trying to get it to work. If you think I should add a different powersupply, I can try that later today.

RonnyRusten, 2019-11-14:
I tried connecting a better power supply, but the settings will not stick. After restart (when clicking "Submit Query") all settings are back to default (also wifi). I have ordered a esp8266 and I will try again when it arrives.

vitotai, 2019-11-14:
Try formatting SPIFFS by
http://[IP of BPL or brewpiless.local:8008/format-spiffs

RonnyRusten, 2019-11-14:
OK, wil try that when I'm home from work. Thank you.

RonnyRusten, 2019-11-14:
Hi, again. This did not make any difference. Still not possible to change the title (I haven't tried anything else). Do you have any other tricks I could try, or should I just wait for my new boards? It will probably take a few weeks, ordered them from China (to Norway) a few days ago.

vitotai, 2019-11-16:
try access the setting directly by 
http://[IP of BPL or brewpiless.local:8008/filemanager
you should find a file name "bpl.cfg".

RonnyRusten, 2019-11-17:
There are no files on that address, see attached screendump.
[image]

RonnyRusten, 2019-11-18:
Hi, again.
I'm sorry, I just found out how PlatformIo works...
When I uploaded the env:classic or env:newui tasks, it works as it should.
The "default" tasks seems to be for the d1pro (or something?)
Sorry for being "stupid" (but maybe this should be explained somewhere in the wiki?)
Thank you for trying to help me, I actually learned a few things in the process.

vitotai, 2019-11-22:
Glad you found the problem.
I constantly change the platform.ini. Therefore, the ability to understand and modify platformio.ini seems to be necessary for those who want to build the project.

closed 2019-11-24 by RonnyRusten