vitotai/BrewPiLess #201: Set Original Gravity and Add Gravity Record

RonnyRusten, 2019-11-24:
Is there a bug in the manual gravity input dialogs? If I type in a value, it shows it as a different value at 0C and NaN@.C
See attached image.
(I'm a beer brewing newbie, so there might be a good explanaition to the values...)
[image]

vitotai, 2019-11-26:
The first ~ value shows the gravity reading at "calibrating temperature", which can be set in "Gravity" pages.
The seance ~ value shows the gravity reading at current Beer temperature, which seems not available in your case.

RonnyRusten, 2019-11-26:
Aha, OK. Then it might give more sense when I get my temperature sensors connected, and I will set the calibrating temperature to 20 degrees in the gravity page. Thank you.

closed 2019-11-27 by RonnyRusten