vitotai/BrewPiLess #202: Method to reset graph so it only shows specific dates

sedgington, 2019-11-25:
Hi Vito,
I'm reacquainting myself with this project and am super impressed in the progress that has been made since 2017. I've been building a new brewpiless and testing it, as well as changing it so it will work with a glycol chiller (home made) and as a temp sensor in iSpindel. Now that I'm almost ready to brew with it, I have encountered the frustrating problem that I can't find where to set the date range. I can see how you can pull on specific dates to enlarge the view, but I want to get rid of everything on the graph that I have used for testing. I can't seem to find the solution although I seem to remember there is one. I erased the eprom, obviously updated to the latest version, even went so far as to lengthen the time for the external temp sensor from 300 to 900 in visual studio, but ithe data doggedly stays there on the graph. I know this is going to be simple, but I can't find the answer anywhere on google. Could you please let me know what I'm doing wrong?
Also, have you done anything more on using the internal temp sensor on the iSpindel as the beer temp sensor. Increasing the time to 900 improved it (as you had suggested , ibelieve in #98) but sometimes I still don't get a stable reading. With Brewfather, it wants a 900s polling time as the minimum value, so I moved it to 900 on brewpiless in order that it wouldn't see the sensor as dead. Do you think making this polling time might be better as the iSpindel only fires every 900s to save battery. I get no problem with the tilt, just the external temp sensor. 
In advance, thanks for any help you can provide
[image]

sedgington, 2019-11-25:
Well, as usual, as soon as you ask the question in public you figure out the answer. Apparently I had started logging data some time ago and forgot about it, once, I went there and stopped logging, the graph only showed current data. Hope this helps someone else.

However, any update on timiing for iSpindel temperature sensor would be appreciated deeply.

Hammersoe, 2019-12-07:
Hi!
I am curious about the gluycol chiller temp control you have added- what features are you getting in advance of just simply using the output for the "Chamber cooler" or "Chamber heater" to activate the pump for circulation? Have you added any items under setup or config as well?

sedgington, 2019-12-10:
I just followed the documentation  to control the glycol chiller cooling set point:
https://github.com/vitotai/BrewPiLess/blob/master/doc/glycol.md
and here
https://github.com/vitotai/BrewPiLess/blob/master/doc/ParasiteTemperatureControl.md
Works great

closed 2019-12-20 by sedgington