vitotai/BrewPiLess #205: MQTT Remote Control

jbishop129, 2019-12-10:
Hello, Great project!!
I cannot seem to get MQTT Remote Control to work. I'm able to get it to attach to my MQTT broker and see the logging in topics as expected, both when set as Individual Topic or JSON; working, looks great. However, I cannot seem to get it to switch modes, or update fridge or beer set points via MQTT. I suspect I'm missing something in the topics I'm sending those messages. Here's my setup:

Base Path: `BPL01`
Path for Mode: `setMode`
path for FridgeSet: `setFridge`
path for BeerSet: `setBeer`

I've tried sending any of the `0,1,2,3` payloads, or `o,p,f,b` payloads to topic:
`BPL01/setMode` topic. I see the payload publish in the topic as expected when monitoring my broker, but no mode change.
I've also tried prefixing the Base Path with `/`,  and targeted the topic accordingly (`/BPL01/setMode`), and also tried prefixing the Path for Mode: path with a `/`, just in case it wasn't automatically assuming a `/` between the base and topic, but to no avail.

Worth mentioning, setting setpoints and modes via the web interface is working fine, and in that perspective, the PID controller is working just dandy, controlling my keezer and heating element, reporting the data back to MQTT topics, even got my iSpindel working in there too. 

So with that, 2 questions: What dumbly obvious thing am I missing, and second, is there logging I can review on the device to see where I've gone wrong for myself?

Thanks!

vitotai, 2019-12-11:
First of all, starting "/" is necessary. (I might add extra code to ensure that in next release.)

I ran out of ubidots credits. The only server I can test is CloudMQTT. It works great for me. 
Maybe, I say Maybe. Did you set "MQTT Remote Control" enabled to ON? The switch is at right most position, and sometimes I forget to check it.

jbishop129, 2019-12-11:
I tried adding the preceding `/` to each of the remote control endpoints, and verified the `ON` slider is selected, still same symptoms. I'm experiencing another issue that I now suspect is related; this is repeatable on multiple devices with the oled firmware at least - here's the steps to reproduce:
In WebUI:
* Go to `Control` tab
  * select either Beer Const, Fridge Const tabs
  * enter a temperature set point, for example `68` (reminder: I'm running i Fahrenheit mode)
  * click Apply
  * Error! I get a pop-up: `Not connected to controller`
  * second popup: `Invalid Temperature:68`

Workaround:
* Go to Beer Profile tab
* Apply whatever profile is in there, doesn't seem to matter what it is
* Error! same pop-up: `Not connected to controller`
* second pop-up asking for Original Gravity, defaulted to 1.0.
  * click either `OK, just Go` or `Set OG, and Go`
!! Without changing to any other tab in the UI, staying on the Control tab !!
* Now go to either Beer Const or Fridge Const
* The `68` temp I tried to set earlier is already populated
* Click Apply -- Success! Navigating back to the Graph tab, i see it is in the set mode, at the right temp, and the controller does its' job properly from there. 

Workaround 2:
* Same as above, but set the mode to `Off`
* When clicking Apply in the Off mode, it pops up the same: `Not connected to controller`, however it *does* actually set the mode to Off successfully regardless of this msg.
* Now I can navigate to either Beer Const or Fridge Const, set a temp, and successfully apply it. Again, you must not navigate out of the Control section of the UI after setting Off mode and before applying a temp in Beer/Fridge Const modes, or it wont take. 

Worth reiterating: Now that I've navigated away from the Control tab, if I want to go set the temp setpoint higher/lower, or change a different mode (Beer Const, Fridge Const), I have to go thru the workarounds described above. 

Hopefully not overly verbose here :) Any thoughts on this issue and if it is related to the MQTT issue?

vitotai, 2019-12-12:
Please update too v4.0. It's might be easier for me.

The issue of "Invalid Temperature:68" has the same cause of "Not connected to controller". For some reason, the controller is not connected, and valid temperature range is not available, nor the temperature unit.

Please use chrome, open Javascript Console, and check the log.

jbishop129, 2019-12-12:
I don't find any v4.0 update, most current I find is 3.6r3; how can I upgrade?

vitotai, 2019-12-12:
It's on another branch.
https://github.com/vitotai/BrewPiLess/tree/4.0pre

jbishop129, 2019-12-12:
Thanks, I updated to `BrewPiLess.v4.0a1.oled.newui.4m2m.bin` successfully, and this resolved the UI issues described above. Unfortunately the MQTT remote control issues still persist; testing with every combination of prefixed `/` and not, both on the base and on the subscribed topics. Is there system logging I can look at, where it might show me something to the effect of precisely what topic it is subscribed to? Perhaps a way to see that in the serial monitor?

vitotai, 2019-12-13:
What MQTT broker are you using? 
There are not many logs related to MQTT because it just worked without much testing.

jbishop129, 2019-12-13:
I'm using mosquitto. I run it as a container, working quite well for a few other IoT services. I'm running it without any auth requirements or TLS, so there shouldnt be any roadblocks in connectivity. I see BPL reporting all sensor data to the broker just fine, not sure why BPL doesn't seem to find the remote control cmds on the subscribe topic. 

I wish there were a way to see in logging or serial monitor, exactly what topic BPL is monitoring, after applying config.

vitotai, 2019-12-13:
Serial logging is enabled by adding "-DSerialDebug=true" to "build_flags". You can reference "esp8266dev" configuration.

shambolic94, 2019-12-14:
For what it's worth, I also run the BPL in Fahrenheit mode and have the exact same issue as the OP getting the "not connected to controller" and "invalid temperature" errors.  The same workarounds work for me.  In addition, I've discovered that if I re-enter the command for switching to Fahrenheit on the testcmd page, I can adjust temps for a short time without getting the error message.

Not sure if it's related, but I've also found that entry of OG and SG seems broken in Fahrenheit mode.  I've found that to get "Gravity Data" to correctly display my OG or SG, I need to input a temperature of 138 degrees.  I'm guessing it's not coincidental that 138°F = 59°C

I am running BrewPiLess.v4.0a1.thorrax.newui.4m2m.

shambolic94, 2019-12-14:
One more thing I forgot, which further suggests that this is an issue with fahrenheit mode:  If I input a beer constant of "20," it will successfully set the BPL to 20°F (although I still get a "not connected to controller" error first).  

So, in other words, it refuses to set a beer temp of 68°F (or other temps in that range) without using one of the various workarounds, but will happily set a beer temp of 20°F with no need for workarounds.

vitotai, 2019-12-18:
The DEFAULT temperature unit and ranges are in Celsius. Setting temperature out of ranges will be rejected.
If the browser does not connect to BPL correctly, which is the root cause as I mentioned, DEFAULT settings are applied.

Open the Javascript console to see what happens.

shambolic94, 2019-12-18:
Here is what the Javascript console shows when I navigate to the control.htm page: 

[image]

vitotai, 2019-12-18:
It is a bug caused by timing issue. Will update later. (I need to study my code first.)

shambolic94, 2019-12-19:
That seemed to fix it.  Thanks!!

jbishop129, 2019-12-20:
+1 for the `BrewPiLess.v4.0a2.oled.newui.4m2m.bin` having fixed the described workarounds for setting the fridge setpoints, thanks!! Updating mode or setpoints via MQTT issue still persists. @shambolic94 any chance you use MQTT, and have the ability to test remote control functionality? If so and if working for you, I'd love to know precisely what your config is, and the topic(s) you publish to, controlling the BPL

vitotai, 2019-12-23:
I've said that I used ubidots.com as illustrated in the document.
Now I can only test on cloudmqtt.com.

shambolic94, 2019-12-26:
jbishop129, I have used ubidots in the past on previous builds and it worked great.  I think I just followed the instructions in the documentation on github and it worked on first try.  Just tried to re-enable it, and have found that the "submit" button on the logging screen in BPL no longer works.  Console produces this error:  
[image]

jbishop129, 2019-12-26:
@shambolic94 thanks for the feedback, and I also see this newly reported
bug. I thought it was just me, but I noticed all MQTT settings are blanked
out in the config, and submitting just zeros them all out again. MQTT is
still reporting to the msg topics as I had it configured before the latest
update, but it seems the ability to update has broken, and your console
screenshot seems to show the culprit.

vitotai, 2019-12-27:
It seems like I messed up. Will fix when I have time.

konradmt, 2021-01-18:
@jbishop129 Were you able to get temp updated via MQTT? 

I have the same setup as you, except I'm using BPL 4.1: D1Mini 8266, MQTT running as a container inside HA. 

I can change temps via web interface. When I send a MQTT message to change the temp, it gets reverted to what I set in the web interface.

jacyker, 2021-01-18:
@konradmt Currently, control by MQTT does not work (stopped working after one of the "recent" 4+ updates). Eg. the sent temperature or mode goes to the MQTT channel, but after a while it is "overwritten" by the temperature or mode set by the WWW. I have recently reported: https://github.com/vitotai/BrewPiLess/issues/284

jalim, 2023-05-20:
I haven't had any luck publishing commands via MQTT to Brewpiless, has this issue ever been resolved?

denisfrench, 2024-10-10:
For anyone else struggling with this, be aware the "MQTT remote control" topics ignore the ```Base Path``` set within "Reporting", so you need to specify the complete topic path.

Setting ```Path for mode``` to ```brewpiless/modeControl``` is now working for me on https://github.com/vitotai/BrewPiLess/releases/download/4.4/BrewPiLess.v4.4.esp32.bin 

Unfortunately when mode or the setpoints are changed the MQTT reporting is not updated straight away, instead waiting until the next ```Publish Period``` to update. So there is no immediate feedback via MQTT (but there is in the UI).